sonocotta/esp32-audio-dock #4: Loosing connection

agdamsbo, 2024-03-20:
Hi. I have purchased your Louder ESP board on tindie. I find that the connection is often lost to my LMS server during playback. Do you have any suggestions for improving? I haven't been able to find alternative mini ESP32 board with external antenna. 

I run squeezelite-esp32 on the unit, its build into an old speaker with wooden cabinet. Stream is from the Music Assistant (LMS functinality) for Home Assistant.

anabolyc, 2024-03-20:
Hey @agdamsbo 

Most probably this relates to connectivity quality. Sometimes you can improve by taking the device closer to the WiFi router/repeater or changing your wifi channel. An alternative way is to use Ethernet instead. It is a plug-and-play solution, but you need to have an ethernet wire around. 

If you suspect a specific ESP32 module to be defective, you can run some Arduino samples with a wifi scan and compare it to a known good module.

agdamsbo, 2024-03-20:
Hi. Thanks. The speaker sits next to my router, forgot to mention. Will try an other board. And might consider [adding an external antenna like here](https://community.home-assistant.io/t/how-to-add-an-external-antenna-to-an-esp-board/131601/38)

anabolyc, 2024-03-20:
@agdamsbo I'm working on [next-generation louder esp](https://hackaday.io/project/194329-louder-esparagus) now. Please subscribe to [my campaign](https://www.crowdsupply.com/sonocotta/esparagus-media-center) to do not miss when it will be out.

agdamsbo, 2024-03-20:
Yes, I have seen that, but I have also recently bought 3 esp louder boards, And I don't want them to be wasted. Also have the Espargus Hifi medialink and it seems to work great.

Just had a look at my router, and the board is reported as having an excellent connection.

anabolyc, 2024-03-21:
Please keep me posted. Also interested if snapcast would work better

closed 2024-05-06 by anabolyc