sonocotta/esp32-audio-dock #9: Flashing snapclient on Louder ESP rev E error

agdamsbo, 2024-06-19:
Hi. I tried your webflasher to flash snapclient to one of my Louder ESP rev E boards. I get the following error, which sends the board into a rebooting loop:

```
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7428
load:0x40078000,len:16652
load:0x40080400,len:4
load:0x40080404,len:4284
entry 0x40080668
I (29) boot: ESP-IDF v5.1.1 2nd stage bootloader
I (29) boot: compile time Jun 14 2024 20:30:26
I (29) boot: Multicore bootloader
I (33) boot: chip revision: v3.1
I (37) qio_mode: Enabling default flash chip QIO
I (43) boot.esp32: SPI Speed      : 80MHz
I (47) boot.esp32: SPI Mode       : QIO
I (52) boot.esp32: SPI Flash Size : 4MB
I (56) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (80) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (88) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  3 ota_0            OTA app          00 10 00010000 001b0000
I (102) boot:  4 ota_1            OTA app          00 11 001c0000 001b0000
I (110) boot:  5 storage          Unknown data     01 82 00370000 00090000
I (118) boot: End of partition table
I (122) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=341c8h (213448) map
I (189) esp_image: segment 1: paddr=000441f0 vaddr=3ffb0000 size=03740h ( 14144) load
I (194) esp_image: segment 2: paddr=00047938 vaddr=40080000 size=086e0h ( 34528) load
I (206) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=cdde0h (843232) map
I (438) esp_image: segment 4: paddr=0011de08 vaddr=400886e0 size=0c6cch ( 50892) load
I (466) boot: Loaded app from partition at offset 0x10000
I (466) boot: Disabling RNG early entropy source...
I (477) cpu_start: Multicore app
I (477) quad_psram: This chip is ESP32-D0WD
I (478) esp_psram: Found 8MB PSRAM device
I (479) esp_psram: Speed: 80MHz
I (483) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (491) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (500) cpu_start: Pro cpu up.
I (504) cpu_start: Starting app cpu, entry point is 0x400814f4
I (496) cpu_start: App cpu up.
I (978) esp_psram: SPI SRAM memory test OK
I (986) cpu_start: Pro cpu start user code
I (986) cpu_start: cpu freq: 240000000 Hz
I (986) cpu_start: Application information:
I (989) cpu_start: Project name:     snapclient
I (994) cpu_start: App version:      1
I (998) cpu_start: Compile time:     Jun 14 2024 20:30:59
I (1004) cpu_start: ELF file SHA256:  533ea0f74923bc30...
I (1010) cpu_start: ESP-IDF:          v5.1.1
I (1015) cpu_start: Min chip rev:     v0.0
I (1020) cpu_start: Max chip rev:     v3.99 
I (1025) cpu_start: Chip rev:         v3.1
I (1030) heap_init: Initializing. RAM available for dynamic allocation:
I (1037) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1043) heap_init: At 3FFB9EF8 len 00026108 (152 KiB): DRAM
I (1050) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1056) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1063) heap_init: At 40094DAC len 0000B254 (44 KiB): IRAM
I (1069) esp_psram: Adding pool of 4096K of PSRAM memory to heap allocator
I (1077) spi_flash: detected chip: generic
I (1081) spi_flash: flash io: qio
I (1086) app_start: Starting scheduler on CPU0
I (1090) app_start: Starting scheduler on CPU1
I (1090) main_task: Started on CPU0
I (1100) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1108) main_task: Calling app_main()
I (1123) SC: Start codec chip
I (1123) HAL: INIT
W (1123) TAS5805M: Power down pin: 33
W (1144) TAS5805M: Setting to HI Z
E (1144) TAS5805M: Fehler bei der I2C-Übertragung: ESP_FAIL
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40104189
file: "./components/custom_board/tas5805m/tas5805m.c" line 162
func: tas5805m_init
expression: tas5805m_write_byte(TAS5805M_DEVICE_CTRL_2_REGISTER, 0x02)

abort() was called at PC 0x4008d30f on core 0

Backtrace: 0x40081dd9:0x3ffbb5e0 0x4008d319:0x3ffbb600 0x40091689:0x3ffbb620 0x4008d30f:0x3ffbb690 0x40104189:0x3ffbb6c0 0x401042cc:0x3ffbb700 0x40103d7d:0x3ffbb730 0x40103e30:0x3ffbb780 0x400da95a:0x3ffbb7b0 0x40199a5e:0x3ffbb810

ELF file SHA256: 533ea0f74923bc30

Rebooting...
ets Jul 29 2019 12:21:46
```

anabolyc, 2024-06-19:
Hey @agdamsbo Can you please attach a photo of your setup. Looks like I2C communication fails for some reason.

agdamsbo, 2024-06-19:
I have the espboard attached to the carrier ESP louder board, microusb from espboard to my computer (macos). I will try again with a different computer maybe if you think its just a communications thing.

anabolyc, 2024-06-19:
It is a communication between module and the DAC that is failing, that's why I'd like to see an actual photo to verify it everything is correct

agdamsbo, 2024-06-19:
[image]

anabolyc, 2024-06-19:
@agdamsbo in the Revision E DAC cannot pull power from USB (Few customers reported earlier that it blows power circuit on the module). Therefore you need to connect power source to the DAC using barrel jack or screw terminals, keeping USB power for flashing and debugging. Right now your DAC is not powered, as indicated by LED near the barrel jack.

Staring Rev H, USB can power DAC as well, since I changed ESP32-mini for ESP32-WROVER module.

agdamsbo, 2024-07-01:
Hi. So now I finally have a little time on hand to revisit this issue. I've had a look at my board again. There is one capacitor (I assume) that looks to be shortening the connection on the mini32 (red circle). And there is another capacitor (again, just my assumption) on the DAC, that is not properly soldered to the pad (yellow circle). I have just marked on the same photo, as uploaded previously. I assume these might be issues (especially on the DAC), that might cause the bad reception I have experienced? Thank you in advance.
[image]

anabolyc, 2024-07-01:
Hey @agdamsbo good catch! DAC capacitor is just one out of many filtering power, so it should not cause malfunction. If it were shorted, it would cut out power altogether, so probably it is not soldered. You may solder it back in, if you have soldering iron, or just leave it as it is. 

As for the module itself, it is hard to say what it is doing, but I assume it is filtering power. It can affect the network, although that would be quite exotic I'd say. Again, if you have a soldering iron, I'd advise to attempt solder it back in, or remove it altogether and see if it helps. Not much to lose anyway :)

closed 2024-12-17 by anabolyc