sonocotta/esp32-audio-dock #14: boot_comm: mismatch chip ID, expected 9, found 0 -- when ota louderesp32

olivervhansen, 2024-08-20:
I'll admit I'm a little unsure of which device I have but it was a recent purchase so I believe rev J of the WROOM (that's what the chip says on the board).

I'm installing ESPHome from Home Assistant when I get this error. I can get the base install of ESPHome from web.esphome.io to install and even get it into Home Assistant after but when I try to update the config to  [https://github.com/sonocotta/esp32-audio-dock/blob/main/firmware/esphome/louder-esp32-media-player.yaml](https://github.com/sonocotta/esp32-audio-dock/blob/main/firmware/esphome/louder-esp32-media-player.yaml) I get the error in the subject.

```[20:33:17][D][esphome.ota:293]: Progress: 78.2%
[20:33:18][D][esphome.ota:293]: Progress: 90.5%
[20:33:19][D][esphome.ota:293]: Progress: 98.1%
[20:33:20][D][esp-idf:000]: E (2370958) boot_comm: mismatch chip ID, expected 9, found 0

[20:33:20][E][ota.arduino_esp32:052]: End error: 9
[20:33:20][W][esphome.ota:309]: Error ending update! error_code: 132
[20:33:20][E][component:164]: Component esphome.ota set Error flag: unspecified
[20:33:20][W][component:237]: Component esphome.ota took a long time for an operation (9670 ms).
[20:33:20][W][component:238]: Components should block for at most 30 ms.
[20:33:25][E][component:176]: Component esphome.ota cleared Error flag
```
I've been trying lots of things but this is the latest config I'm using that fails. I'm guessing it's probably something simple. Searching this specific error gives me results about encryption so I've tried installing without it.

```
substitutions:
  name: "louder-esp32"
  friendly_name: Louder-ESP32 Media Player
  pcb_version: "J"

esphome:
  name: "${name}"
  name_add_mac_suffix: false
  comment: "${friendly_name} Louder ESP32 ${pcb_version}"
  includes:
    - louderesp32.h
  platformio_options:
    lib_deps: "Wire"
  on_boot:
    priority: 800
    then:
    ## Set a volume limit just to be safe...
    - media_player.volume_set:
        id: louderesp32
        volume: 20%

esp32:
  board: mhetesp32minikit
  #framework:
    #type: arduino

### Optional if you want ethernet (then remove all wifi config) ###
#ethernet:
#  type: W5500
#  clk_pin: GPIO18
#  mosi_pin: GPIO23
#  miso_pin: GPIO19
#  cs_pin: GPIO05
#  interrupt_pin: GPIO35
#  reset_pin: GPIO14

# Enable logging
logger:
  level: DEBUG
  
# Enable Home Assistant API
api:
    #encryption: 
      #key: !secret esphome_api_key

# Allow Over-The-Air updates
ota:
  - platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret wifi_domain
  ap:
    ssid: "$name Hotspot"
    password: "squeezebox"

captive_portal:

psram:
  mode: octal
  speed: 80MHz

switch:
  - platform: custom
    lambda: |-
      auto tas5805 = new TAS5805();
      App.register_component(tas5805);
      return {tas5805};
    switches:
      name: "Enable Amp"

i2c:
  sda: GPIO21
  scl: GPIO27
  scan: True
  id: i2c_component

i2s_audio:
  i2s_lrclk_pin: GPIO25
  i2s_bclk_pin: GPIO26

media_player:
  - platform: i2s_audio
    name: $friendly_name
    id: louderesp32
    dac_type: external
    i2s_dout_pin: GPIO22
    mode: stereo
```

anabolyc, 2024-08-20:
Hey @olivervhansen 

Rev J is an ESP32-S3 board, this config is for the regular ESP32. Two things need to be changed: the board and the pinout.
You can find pinout [here](https://github.com/sonocotta/esp32-audio-dock/tree/main#louder-esp) and board should be `esp32-s3-devkitc-1` afaik. 

Pls give it a try, and I can add updated config to the repo

olivervhansen, 2024-08-20:
Thanks @anabolyc ! Those were the pieces of information I didn't have. I had a config in HA that worked using that board but then it gave me errors about the pins and because I didn't know much about the pins I figured that file in the repo must be correct.

I added the encryption back in after one successful flash and this is what I currently have working. It loads but I don't have my speakers yet to test it fully.

One other question: is **pcb_version** supposed to be defined elsewhere? I added it to substitutions but initially thought it should be pulled in from something that detects the version.

```
substitutions:
  name: "louder-esp32"
  friendly_name: Louder-ESP32 Media Player
  pcb_version: "J"

esphome:
  name: "${name}"
  name_add_mac_suffix: false
  comment: "${friendly_name} Louder ESP32 ${pcb_version}"
  includes:
    - louderesp32.h
  platformio_options:
    lib_deps: "Wire"
  on_boot:
    priority: 800
    then:
    ## Set a volume limit just to be safe...
    - media_player.volume_set:
        id: louderesp32
        volume: 20%

esp32:
  board: esp32-s3-devkitc-1

### Optional if you want ethernet (then remove all wifi config) ###
#ethernet:
#  type: W5500
#  clk_pin: GPIO18
#  mosi_pin: GPIO23
#  miso_pin: GPIO19
#  cs_pin: GPIO05
#  interrupt_pin: GPIO35
#  reset_pin: GPIO14

# Enable logging
logger:
  level: DEBUG
  
# Enable Home Assistant API
api:
    encryption: 
      key: !secret esphome_api_key

# Allow Over-The-Air updates
ota:
  - platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret wifi_domain
  ap:
    ssid: "$name Hotspot"
    password: "squeezebox"

captive_portal:

psram:
  mode: octal
  speed: 80MHz

switch:
  - platform: custom
    lambda: |-
      auto tas5805 = new TAS5805();
      App.register_component(tas5805);
      return {tas5805};
    switches:
      name: "Enable Amp"

i2c:
  sda: GPIO8
  scl: GPIO9
  scan: True
  id: i2c_component

i2s_audio:
  i2s_lrclk_pin: GPIO15
  i2s_bclk_pin: GPIO14

media_player:
  - platform: i2s_audio
    name: $friendly_name
    id: louderesp32
    dac_type: external
    i2s_dout_pin: GPIO16
    mode: stereo
```

anabolyc, 2024-08-20:
**pcb_version** is completely optional. It is just a handy way to remind the board it was tested with.

olivervhansen, 2024-08-21:
So this config works to flash the device but I'm not getting expected behavior in HA. If I manually turn on the "Enable Amp" switch  under ESPHome it changes back to off. If I do it again it stays on for a while but then seems to automatically turn off. I can try to play media with it on or off and nothing happens. I've tried from Music Assistant and also just sending a TTS command from a local piper engine. I connected one passive speaker I had for testing to the left channel only and nothing.

A portion of the logs
```
Louder-ESP32 Media Player changed to Idle Media player: Play media
5:11:31 PM - 15 hours ago - Supervisor
Enable Amp turned off
5:11:31 PM - 15 hours ago
Enable Amp became unavailable
5:11:31 PM - 15 hours ago
Louder-ESP32 Media Player became unavailable Media player: Play media
5:11:31 PM - 15 hours ago - Supervisor
Enable Amp turned on Switch: Turn on
5:10:32 PM - 15 hours ago - Admin
```

anabolyc, 2024-08-21:
Can you please enable DEBUG level of logging on the ESPHome and attach the logs from the louder boards when you switch the amp on

olivervhansen, 2024-08-21:
Thanks for the suggestion! I wasn't sure how to enable debug but once I did I could see what was going on. My HA is on a different VLAN than the media players and this one hadn't yet been added to the list of devices that could initiate traffic to the HA. So HA could reach out to Louder-ESP32-S3 but Louder couldn't initiate a session or grab the audio file generated on the HA for the TTS.

Fixed that and now I've got sound coming out to my test speaker now so I think it's functioning properly now.

anabolyc, 2024-08-22:
Thanks for letting me know. I added S3 config to the repo to make it easier next time.

closed 2024-08-22 by anabolyc