sonocotta/esp32-audio-dock #15: Speaker Connection for Mono/PTBL mode

flix-one, 2024-09-05:
Is it possible that the images for the speaker connection are switched accidently between BTL and PBTL modes?
The schematics and rated power fit together but the speaker connection images are showing the opposite of the schematics?

https://github.com/sonocotta/esp32-audio-dock#btl-and-pbtl-mode

To use the Louder ESP32 rev. H with one speaker, I need to solder the mono header together horizontally and then connect OutA+ and OutA- to Speaker + and Out B+ and OutB- to Speaker - (As shown in schematic for PTBL but image for BTL)
Is that correct?

anabolyc, 2024-09-05:
Hey @flix-one you're right, thanks for reporting. I've updated the description, and also added a sentence about the software side of that.

closed 2024-09-05 by flix-one

flix-cas, 2024-09-10:
Just a quick clarification for the PTBL mode.
When I solder the mono header together, is it sufficient to connect the Speaker+ to **either** OutA+ **or** OutA- ?
Or do I have to bridge the OutA+ and OutA- together before connecting Speaker+ as in your [picture](https://github.com/sonocotta/esp32-audio-dock/assets/5459747/eca2cf1e-ebe0-4684-9aa2-1cc193e0982f)?

anabolyc, 2024-09-10:
> Just a quick clarification for the PTBL mode. When I solder the mono header together, is it sufficient to connect the Speaker+ to **either** OutA+ **or** OutA- ? Or do I have to bridge the OutA+ and [...]

Correct, you can use either, they are bridged together

flix-one, 2024-09-10:
Sorry wrong account, but thank you for the super fast reply :-)