sonocotta/esp32-audio-dock #16: Add rotary encoder to Louder ESP32 (rev H)

flix-one, 2024-09-08:
I really love the Louder ESP32, it works really well.
I would like to add a rotary encoder to add the possibility to change volume directly at the speaker.

Could you give some information how this is done easiest?
I've seen some GPIOs available on the PCB of my rev H (2, 4, 13, 15, 32, 36, 3V3, GND).
Can I use all of them or are some of them better suited for the rotary encoder?
I suppose the programming needs to be done via the I2C or SPI Bus Parameters in the Hardware Section of Squeezelite-ESP32?

I have some KY-040 available here, but it seems they're working with 5V?
https://www.az-delivery.de/en/products/drehimpulsgeber-modul
"GND"
"+" = 5V
"SW" = Interrupt for Push-Button
"DT" = Encoder Pin B
"CLK" = Encoder Pin A

If they're not suited, could you give me a suggestion which ones are better suited?

Thanks in advance, maybe this could also be added to the readme?

anabolyc, 2024-09-08:
Hey @flix-one 

First of all, rotaries are not active devices, they are basically switch whatever voltage you feed in, like basic buttons. Please use 3.3V as power and you'll get good reliable 3.3v on all signal lines.

You can use pretty much any pins that are available with [some limitations](https://randomnerdtutorials.com/esp32-pinout-reference-gpios/). In your list avoid pin 2 (bootstrapping pin), the rest should be good to go.

Last point, here is the [squeezelite](https://github.com/sle118/squeezelite-esp32?tab=readme-ov-file#rotary-encoder) docs on how to configure rotary encoder as soon as you have it connected.

flix-one, 2024-09-08:
Thank you for the feedback, I think of using 
SW = GPIO 4
CLK (Pin A) = GPIO 13
DT (Pin B) = GPIO 32

In the Pinout Reference it says:

```Note: ADC2 pins cannot be used when Wi-Fi is used. So, if you’re using Wi-Fi and you’re having trouble getting the value from an ADC2 GPIO, you may consider using an ADC1 GPIO instead. That should solve your problem.```

Are the ADC2 pins (0, 2, **4**, 12, **13**, 14, 15, 25, 26, 27) used for Wi-Fi by your PCB configuration (especially 4 and 13)?

anabolyc, 2024-09-08:
As far as I know, this warning only applies if you're using ADC, or `analogRead`. For digital functions it is fully functional.

flix-one, 2024-09-09:
With SW on GPIO 4, DT (A) on GPIO 32 and CLK (B) on GPIO 13 it works with rotary_config: ```A=32,B=13,SW=4,volume,longpress```

Thank you for the support!

closed 2024-09-09 by flix-one

anabolyc, 2024-09-09:
You're very welcome. Feel free to attach a pic of the end result, if it is all right :)

flix-one, 2024-09-10:
[image]
[image]