sonocotta/esp32-audio-dock #19: [Idea] Power Supply via PoE

flix-one, 2024-11-21:
Hi,

I was asking myself if it would be possible to use the Ethernet connection to supply the Louder-ESP32 with power via PoE connection. 

One problem is probably the voltage level of 48V that has to be reduced to the 12-18V of the ESP32?
I found the TI TPS23730 that can convert DC-DC from PoE
https://www.ti.com/product/TPS23730

Then you would probably have to redesign the PCB to first extract the power supply from the Ethernet cable and then go through the Ethernet module to connect the data stream?

It's just an idea, but the benefit would be to supply a speaker with data and power by only one ethernet cable and have wired connection. This could be beneficial in case WiFi coverage is bad at the destined location of the speaker.

Greetings,
Fabian

anabolyc, 2024-11-21:
Hey @flix-one 

A few more people reached out to me with a similar request. I'm not against it, but I would set my expectations low for a few reasons
- PoE appliance can deliver 50W only, which could not be enough for the DAC
- Power quality on the PoE line is far from audio grade, the D-class amp will push all the noises directly to the speakers
- Long power lines together with high-speed digital lines don't sound good for audio either

With that said I'm not against it and I'm planning to do a prototype at some point. For now, I'd need to investigate how the PoE works, and what should be done on the ESP side to negotiate voltage and drop it down to 3.3V (I guess it is similar to USB-PD?). I also need to find some PoE appliances for testing, since I don't have them to start with :)

bitte-ein-bit, 2025-01-16:
For a start, you can use POE splitters. They receive POE+DATA and split that into DATA (RJ45) and Power (usb, barrel, open ends). The voltage depends on the splitter chosen.

To power it a simple POE injector might be sufficient.

One item to be aware of are the two flavors: active and passive POE. Passive always gets power (various voltages), active has to negotiate first but is always 48V (to my knowledge)

closed 2025-04-09 by anabolyc