sonocotta/esp32-audio-dock #23: Louder doen't seem to remember wifi details

joep2000, 2025-01-12:
ESP-32 Louder doesn't seem to remember the wifi details after initial flashing.
Configuration is done in recovery mode @ 192.168.4.1 
Initially Telnet is not enabled, however, after connecting to a (hidden) netwerk I get following output (via serial)

`I (454) esp_image: segment 3: paddr=0035b694 vaddr=3fc97cESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42006e08
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1820
load:0x403c9700,len:0xe80
load:0x403cc700,len:0x3110
entry 0x403c9980
I (24) boot: ESP-IDF v4.4.7 2nd stage bootloader
I (24) boot: compile time 13:57:36
I (25) boot: Multicore bootloader
I (26) boot: chip revision: v0.2
I (30) qio_mode: Enabling default flash chip QIO
I (36) boot.esp32s3: Boot SPI Speed : 80MHz
I (40) boot.esp32s3: SPI Mode       : QIO
I (45) boot.esp32s3: SPI Flash Size : 8MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (74) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  3 recovery         factory app      00 00 00010000 00140000
I (96) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (103) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (111) boot: End of partition table
I (115) esp_image: segment 0: paddr=00150020 vaddr=3c160020 size=af574h (718196) map
I (233) esp_image: segment 1: paddr=001ff59c vaddr=3fc971b0 size=00a7ch (  2684) load
I (234) esp_image: segment 2: paddr=00200020 vaddr=42000020 size=15b66ch (1422956) map
I (454) esp_image: segment 3: paddr=0035b694 vaddr=3fc97c2c size=04560h ( 17760) load
I (458) esp_image: segment 4: paddr=0035fbfc vaddr=40374000 size=131b0h ( 78256) load
I (475) esp_image: segment 5: paddr=00372db4 vaddr=600fe000 size=0002ch (    44) load
I (483) boot: Loaded app from partition at offset 0x150000
I (483) boot: Disabling RNG early entropy source...
I (495) cpu_start: Multicore app
I (496) opi psram: vendor id : 0x0d (AP)
I (496) opi psram: dev id    : 0x02 (generation 3)
I (498) opi psram: density   : 0x03 (64 Mbit)
I (503) opi psram: good-die  : 0x01 (Pass)
I (508) opi psram: Latency   : 0x01 (Fixed)
I (513) opi psram: VCC       : 0x01 (3V)
I (517) opi psram: SRF       : 0x01 (Fast Refresh)
I (523) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (528) opi psram: BurstLen  : 0x01 (32 Byte)
I (533) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (539) opi psram: DriveStrength: 0x00 (1/1)
I (545) MSPI Timing: PSRAM timing tuning index: 4
I (549) spiram: Found 64MBit SPI RAM device
I (554) spiram: SPI RAM mode: sram 80m
I (559) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (566) cpu_start: Pro cpu up.
I (569) cpu_start: Starting app cpu, entry point is 0x40375854
I (0) cpu_start: App cpu up.
I (996) spiram: SPI SRAM memory test OK
I (1004) cpu_start: Pro cpu start user code
I (1005) cpu_start: cpu freq: 240000000
I (1005) cpu_start: Application information:
I (1008) cpu_start: Project name:     Squeezelite-ESP32
I (1013) cpu_start: App version:      local.500.cmake-master
I (1020) cpu_start: Compile time:     Jul 23 2024 20:55:47
I (1026) cpu_start: ELF file SHA256:  0000000000000000...
I (1032) cpu_start: ESP-IDF:          v4.4.7
I (1037) cpu_start: Min chip rev:     v0.0
I (1042) cpu_start: Max chip rev:     v0.99 
I (1047) cpu_start: Chip rev:         v0.2
I (1052) heap_init: Initializing. RAM available for dynamic allocation:
I (1059) heap_init: At 3FCB18F0 len 00037E20 (223 KiB): D/IRAM
I (1065) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DIRAM
I (1072) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (1078) heap_init: At 600FE02C len 00001FD4 (7 KiB): RTCRAM
I (1085) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (1093) spi_flash: detected chip: gd
I (1097) spi_flash: flash io: qio
W (1101) spi_flash: Detected size(16384k) larger than the size in the binary image header(8192k). Using the size in the binary image header.
I (1116) sleep: Configure to isolate all GPIO pins in sleep state
I (1121) sleep: Enable automatic switching of GPIO sleep configuration
I (1129) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1139) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1149) esp_app_main: Reset reason is: 0
I (1159) esp_app_main: Reboot counter=2

I (1159) esp_app_main: Starting app_main
I (1169) nvs_utilities: Initializing flash nvs 
I (1179) nvs_utilities: Initializing nvs partition settings
I (1209) esp_app_main: Setting up telnet.
I (1209) telnet: ***Redirecting log output to telnet`

No network connection, however all logging to telnet.

What's going on here?

joep2000, 2025-01-12:
enabling telnet seems to break things...

anabolyc, 2025-01-12:
Hey @joep2000 

First of all, squeezelite-esp32 support for ESP32-S3 is unofficial and very much experimental. Please have a look at [this thread](https://github.com/sle118/squeezelite-esp32/issues/419).

Second, I seem to make it work more or less stable (without telnet though), and that's the build you found on the [web flasher](https://sonocotta.github.io/esp32-audio-dock/) site.

I just re-flashed my Louder-ESP32S3 using the link above and it managed to go well past the point where your unit stuck. Can you please try to re-flash with the erase flash option and see if anything changes?

[image]

anabolyc, 2025-01-12:
That's a good sign. I didn't manage telnet to work, although I didn't try too hard, as I used it once. If you have any success, let me know.

closed 2025-02-10 by anabolyc