sonocotta/esp32-audio-dock #24: Minimal volume

majuss, 2025-01-12:
I'am using a ESP32 louder board with S3 J board, bought recently.

It works geat so for, I'am using ESPHome and added a few lines C++ to initialize the DAC in mono mode since I'am driving a 10" Klipsch in ceiling speaker.
My only concern is the minimal loudness. My child will hear stuff to fall asleep and as soon as I hit a volume under 5 in Home Assistant the speaker cuts off. But 5 is quite loud at night.
Is there a way to get a quitier minimal volume? I'am driving the amp with 24 V.

anabolyc, 2025-01-12:
Hey @majuss 

Currently, the ESPHome driver doesn't utilize hardware volume, since it was done in a rush as a proof-of-concept mainly (by a esphome community member, not me). [Here](https://github.com/sonocotta/esp32-tas5805m-dac) is the generic TAS5805 driver that is capable of changing hardware volume in quite an amazing granularity. 

Simplest would be just to initialize the volume at lower level at startup. But maybe it is also possible to drive volume using hardware from HA (instead of software as it is now).

majuss, 2025-01-13:
I found a workaround. I'am using music assistant and there I can normalize each media players volume and set a lower db :) thank you for your clarifications! Any timeline when you will add your driver as a PR to ESPHome?

closed 2025-01-13 by majuss

anabolyc, 2025-01-13:
I don't have the skills to do it. But I'm in touch with a few folks who are working on this, and hopefully, it will be done soon.