sonocotta/esp32-audio-dock #25: Local Playback (LMS) is too loud. Device crashes

haldi4803, 2025-01-13:
Howdy,

I've bought two Louder-ESP32 v2 boards after the one in my showerroom broke down. Maybe i should think about waterproof coating.... it gets steamy in there...
using the 20V 65W and 19V 90W it works out of the box, compared to the old version. But i had the same Problem as on the old board, when playing via LMS and local Files (NAS) getting over 90% Volume the device suddenly becomes mute while LMS continues the music. Only way to get it back to output sound is by turning the device off and then starting playback again. The track will begin from 0:00.
I realized that this is only happening when i use Local Tracks via LMS. Bluetooth input is way quieter and even playing a track from Tidal via LMS is less loud.

The only setting for something similar i could find was the player Setting "Default Adjustment for Remote Streams" which is at -5
[image]

But that should not have any influence on Bluetooth right?

Do you think i don't have enough power? Should i try a 26V PSU?
Speakers are 6 Ohm 100W/60W Denon S 420 Surround. because they are really small, still loud, and have decent quality (for their size)

Or do you know any LMS settings that i'm missing?
I think last time i solved this issue by changing the amp gain at boot with I2C commands. But that would also reduce Bluetooth volume.

anabolyc, 2025-01-13:
Hey @haldi4803.

With the mute issue first that comes to mind, at maximum volume the DAC might be overheating, it depends on outside temperature and fresh air access. Can you please check that (this can be fixed by attaching a small heatsink at the back side of the DAC, it is designed to push the head on the board). Alternatively, it is recommended to switch modulation at a voltage above 12v, I can help you configure that as well).

Sorry I'm not an LMS expert, I'd advise you to ask at [LMS forums](https://forums.slimdevices.com/). But if you change the hardware volume at startup it will definitely affect all audio sources equally.

haldi4803, 2025-01-14:
> this can be fixed by attaching a small heatsink at the back side of the DAC, it is designed to push the head on the board)

You mean not on top of the DAC but bottom side in the PCB?
Will test this evening.

anabolyc, 2025-01-14:
Correct, bottom of the PCB. DAC has the power pad on the bottom and grid of vias to pass it on the back side of the board

haldi4803, 2025-01-14:
i doubt that's the issue... barely 50°C hot.
[image]

Tried with my hairdried on Cold mode (which has huricane level windspeeds) to prevent it from overheating. No difference.
Volume 80 works fine without any issue.
As soon as i go 100 it takes about 5-10 seconds and silences.
I can continue Playback with Volume 80 After shuting it down and starting again 

Took a Telnet Log with sDebug.
Maybe something is written there?
https://pastebin.com/KyL8UEGz

But i guess @Sle118 would me more avid in reading those right?

anabolyc, 2025-01-15:
Thanks for the comprehensive feedback. Since it is working most of the time, I'm not sure guys from squeezelite-esp32 would consider it software related.

Also temperature doesn't look like anything remotely dangerous, it should safely go up to 100C. If you get it right, with heatsink or without, it is the same?

I think it is related to relatively high voltage. If you have something under 20V, give it a try. Otherwise I'd propose to try this:

> In section “2.3 Modulation Mode Selection”:
Figure 11 shows PVDD, which is an important factor for thermal design. The efficiency of TAS5805M
decreases with the increase of PVDD. When PVDD > 19 V, use 1SPW rather than BD, considering the
efficiency. 

In summary, the efficiency of 1SPW is higher than BD mode and it is suggested you choose
1SPW mode in large output condition. In addition, a larger load means better efficiency

If you want to trigger Hybrid mode (to see if that helps), it is fairly straightforward in the nvs settings:
dac_controlset: {"init":[{"reg":3,"val":2},{"reg":3,"val":3},{"reg":2,"val":2}],"poweron":[{"reg":3,"val":3}],"poweroff":[{"reg":3,"val":0}]}

Change to {"reg":2,"val":1} for 1SPW mode, see if it is any good. If possible please take temperature measurements along the way.

haldi4803, 2025-01-15:
> Change to `{"reg":2,"val":1}` for 1SPW mode

Just quickly tested it on the 19V 65W PSU And it still crashed.
I bought a 24V 5A Meanwell PDU that should be "good" quality...
Will test on that later this week.

haldi4803, 2025-01-16:
Also happening on 24V.
IMHO even faster than on 20V... so might not really be a problem with the current but more of a voltage regulation overload ?
Sadly i don't think i have any 12V PSU lying around that does more than 1A...

anabolyc, 2025-01-18:
> Also happening on 24V. IMHO even faster than on 20V... so might not really be a problem with the current but more of a voltage regulation overload ? Sadly i don't think i have any 12V PSU lying around that does more than 1A...

I'll try to find a higher voltage PSU and try to reproduce it. I tried it with 19V and at a very high volume I think I experienced something similar. But it recovered itself after less than a minute.

haldi4803, 2025-01-18:
Its happening with bass heavy music. And only on the parts where there is a lot of bass... That's why I assume it's a power issue.

This was my testing tune.
https://youtu.be/pa8i4sNzL_g?si=hN0boCkJgwEnxT7z

koriwi, 2025-02-19:
Do we have any updates here? I experienced the same. Right now it is not really a problem, as it happens at a volume which i almost never listen to, but it would still be good to fix or work around it.

anabolyc, 2025-02-19:
I'm working to bring DSP settings to the ESP32 platform, I want to play with different modulation settings to figure out what causing the high power consumption. Also, I'll port more detailed fault logging, maybe it will spit out something useful.

I can confirm that even on low volume it can get quite hot all of a sudden, yet I don't understand why. Need to do some testing with different DAC settings

elektrinis, 2025-03-05:
This happened to me as well, a couple of times, on louder-esp32s3. Within one second when volume is increased, so likely not heat related.
I also had issues with LMS itself - audio stopped playing after I went through several radio stations. Had to restart LMS.
A separate problem: I think the DAC really needs a heatsink. I have only one speaker connected, and it becomes too hot to touch at higher volume (19V psu). I think it may overheat in stereo configuration.

anabolyc, 2025-03-06:
Give me some time, I'm implementing detailed fault reading from the DAC as long as few other settings that I want to test together with higher power voltage. What I know that this DAC behaved differently when powered with more than 12V, and probably require register tunings. I'm not there yet to define the right settings.

elektrinis, 2025-03-08:
Happened a few more times (at 80% volume). Just to confirm, it's enough to reboot the player via web interface for function to resume, it's not necessary to remove power.

anabolyc, 2025-03-08:
Good sign, probably something that can be fixed via software. @elektrinis what voltage power source are you using?

elektrinis, 2025-03-08:
> Good sign, probably something that can be fixed via software. [@elektrinis](https://github.com/elektrinis) what voltage power source are you using?

19V laptop brick.

haldi4803, 2025-03-08:
> Happened a few more times (at 80% volume).

Okay, thats low... mostly happens around 93-96% Volume in my case. 90% was safe all time.

koriwi, 2025-03-09:
> I'm working to bring DSP settings to the ESP32 platform, I want to play with different modulation settings to figure out what causing the high power consumption. Also, I'll port more detailed fault [...]

Awesome. I backed off from working on the dsp settings for the DAC in the esp-louder and started working on a standalone DSP with the esp32 which works with i2s input/output directly because i also have a subwoofer which does not use an esp-louder but discrete boards. I know its offtopic, but just wanted to let you know why I havent looked into it yet. :)

elektrinis, 2025-05-21:
Any news on the fix?
It's a cool product, I'm using it every day. Just need these small fixes.

anabolyc, 2025-05-21:
Hey @elektrinis, I haven't run detailed power tests yet. For now, I'd advise reducing analog gain view dBs to limit output driver voltage swing. It can be done via an updated startup sequence, which should include 
```
{"reg": 124, "val": XX}
```

`XX` should be something between 0 and 32, I'd recommend increasing it until it gets stable enough.

[image]

anabolyc, 2025-08-24:
Hey folks. I found some time (finally) to run some power tests. I've placed the results [here](https://github.com/sonocotta/esp32-tas5805m-dac#power-measurements-on-louder-esp32-board), [here](https://github.com/sonocotta/esp32-tas5805m-dac#power-consumption-testing) and [here](https://github.com/sonocotta/esp32-audio-dock?tab=readme-ov-file#louder-esp32-and-amped-esp32-power-considerations)

Results that I gathered:
- Switching frequency and BD-frequency have little to no effect on the efficiency.
- 1SPW modulation mode is more efficient (for the cost of some THD degradation), but it is only worth using under higher VCC. I recommend setting it up if you're on 15V or more
- Bare naked DAC on the Louder-ESP32 can only sustain long-term operation at full power VCC=15V (BD-mode) and VCC=16V (1SPW mode). Above that, it needs passive or active cooling.
- In case you want to stay passive and use voltage above 16V, you need to tune down the analog gain to limit output power (if stability is a priority)
- Another solution, as soon as DAC reports OT warning - lower the volume slowly using automation until it disappears. In my tests, between OT warning and OT shutdown, you have around 10 seconds for that. With passive cooling more, clearly.

elektrinis, 2025-08-25:
Wait. So do we know for certain that overtemperature is the reason for this crash? In such case, a small heatsink should help.
In my case it crashed within one second of loud audio, should be not enough time to overheat.
Also how can we handle this crash, reset DAC automatically?

anabolyc, 2025-08-25:
@elektrinis yes, that's what I witnessed. A small heatsink adds thermal inertia, so you have a little more time to read faults and lower the volume, but with an excess of 3-5W heat on this small board, you won't get away that easily. Active cooling will be required for long-term stability.

If you have this issue with your board, I'd ask you to try it out 
- reading faults when shutdown happens
- lower analog gain and see how it changes
- lower VCC and see if that helps

All can be done with esphome firmware

anabolyc, 2025-08-25:
@haldi4803

jdegenstein, 2025-08-26:
I just want to add that I am also seeing this issue but with a slightly different device -- a Seeed XIAO ESP32S3 wired point-to-point with a PCM5102a DAC dev board. This setup is powered solely via USB. I am considering trying to add a relatively large electrolytic to see if it helps.

When my device crashes it becomes unreachable over wifi.