sonocotta/esp32-audio-dock #27: Please confirm changes necessary for MONO

frdfsnlght, 2025-02-03:
I've just received my Louder-ESP order (2 units, Rev H2, pre-flashed with SqueezeLite-ESP32). I've hooked one of them up in standard stereo mode and it works perfectly as expected. I want to hookup the second unit for MONO operation but want to confirm a few things:

1) I need to bridge the two pairs of connectors between the speaker terminals horizontally.
2) I need to change the dac_controlset NVS parameter to:

`{"init":[{"reg":3,"val":2},{"reg":3,"val":3},{"reg":2,"val":4}],"poweron":[{"reg":3,"val":3}],"poweroff":[{"reg":3,"val":0}]}
`

3) Connect a single speaker between the 2 output terminals.

Assuming I have those items correct, I have some questions:

Q1) I assume with the connectors bridged (item 1), I don't need to create a 'Y' connection between the +/- terminals on each speaker output?
Q2) Which channel will be output to the speaker? Just the left, or just the right? Can it be changed?
Q3) Assuming the previous answer is a single channel (L or R), Is it yet possible to get a true mono (L+R)/2 output? I know all about the difficulty getting programming info for the TI DAC, but have you had any success in that regard?

Thanks for help!

anabolyc, 2025-02-03:
Hey @frdfsnlght 
The steps look right.
Q1)  Correct, you'll effectively connect them together with those jumper bridges. Or vice versa, connect both terminal outputs to the speaker and you may omit the jumper bridges.
Q2) It will output a single channel (can't remember which one). 
Q3) Yes, it can be fixed with input mixer adjustments, but that's a considerably longer command set, and I need to test it first. I'm working currently to implement DSP settings for this DAC on the esp32 and this is one of those features I want to bring. I've made it work with Raspberry driver, but the esp32 implementation is still a work in progress.

anabolyc, 2025-02-04:
Here it is (not tested)

```
#define REG_PAGE_SET 0x00
#define REG_BOOK_SET 0x7f
```

```
    { REG_PAGE_SET, 0x00 },
    { REG_BOOK_SET, 0x8c },
    { REG_PAGE_SET, 0x29 },
    { 0x18, 0x00 }, //  Input Mixer Left to left = -6 dB
    { 0x19, 0x40 },
    { 0x1a, 0x26 },
    { 0x1b, 0xe7 },
    { 0x1c, 0x00 }, //  Input Mixer Right to left = -6 dB
    { 0x1d, 0x40 },
    { 0x1e, 0x26 },
    { 0x1f, 0xe7 },
    { 0x20, 0x00 }, //  Input Mixer Left to right = -110 dB
    { 0x21, 0x00 },
    { 0x22, 0x00 },
    { 0x23, 0x00 },
    { 0x24, 0x00 }, //  Input Mixer Right to right = -110 dB
    { 0x25, 0x00 },
    { 0x26, 0x00 },
    { 0x27, 0x00 },
    { REG_PAGE_SET, 0x00 },
    { REG_BOOK_SET, 0x00 },
```

frdfsnlght, 2025-02-04:
Excellent!

Based on what you've provided, it appears that when MONO mode (PBTL) is enabled, the DAC is only using the left input channel and ignoring the right?

If I wanted to just send the left channel at 0dB, would I use a value like (0x00 0x43 0xda 0x60 (4446816 decimal) for registers 0x18 to 0x1b, and 0x00 for the other 3 input mixer sections?

And likewise if I wanted to just send the right channel, I'd use those same values for registers 0x1c to 0x1f, and 0x00 for the other 3 input mixer sections?

And I assume before setting the input mixer registers, I should include the other register settings already in the "init" section of the existing dac_controlset value?

Assuming my assumptions are correct, I'll try this out later tonight. If my testing works as I hope it will, I'll post the final dac_controlset values here.

My ultimate plan is to use 2 of the Louder-ESP units as a stereo pair with each configured for PBTL mono, and one configured to output only left and the other only right. If that doesn't work out well, I'll just send a mono mix (L+R)/2 to both of them.

Thanks for the help so far!

frdfsnlght, 2025-02-04:
I've had a chance to play with this. The first problem I ran into is that the terminals on the board are mislabeled: left and right are swapped. Playing a test file ([like this one](https://www.aoakley.com/articles/2018-08-24-stereo-test.php)) confirms this.

The next problem I ran into was that it seems the comments in the code snippet you provided are a little wrong. My testing shows register at 0x18 (and the 3 following) are for mixing the left input to the RIGHT output, register 0x1c is for mixing the right input to the RIGHT output, register 0x20 mixes left input to LEFT output, and register 0x24 mixes right input to LEFT output. And when I say LEFT or RIGHT output, I mean AFTER swapping the left/right terminal labels.

This gets confusing fast.

Taking all this into account (speakers swapped and my observations about the registers), the test file plays as you'd expect with the default dac_controlset.

Then I tested these:

**Stereo speakers (BTL mode), mono mix (L+R)/2 output to right speaker:**
`{"init":[{"reg":3,"val":2},{"reg":3,"val":3},{"reg":0,"val":0},{"reg":127,"val":140},{"reg":0,"val":41},{"reg":24,"val":[0,64,38,231]},{"reg":28,"val":[0,64,38,231]},{"reg":32,"val":[0,0,0,0]},{"reg":36,"val":[0,0,0,0]},{"reg":0,"val":0},{"reg":127,"val":0}],"poweron":[{"reg":3,"val":3}],"poweroff":[{"reg":3,"val":0}]}
`

**Stereo speakers (BTL mode), right input to right speaker:**
`{"init":[{"reg":3,"val":2},{"reg":3,"val":3},{"reg":0,"val":0},{"reg":127,"val":140},{"reg":0,"val":41},{"reg":24,"val":[0,67,218,96]},{"reg":28,"val":[0,0,0,0]},{"reg":32,"val":[0,0,0,0]},{"reg":36,"val":[0,0,0,0]},{"reg":0,"val":0},{"reg":127,"val":0}],"poweron":[{"reg":3,"val":3}],"poweroff":[{"reg":3,"val":0}]}
`

**Stereo speakers (BTL mode), left input to right speaker:**
{"init":[{"reg":3,"val":2},{"reg":3,"val":3},{"reg":0,"val":0},{"reg":127,"val":140},{"reg":0,"val":41},{"reg":24,"val":[0,0,0,0]},{"reg":28,"val":[0,67,218,96]},{"reg":32,"val":[0,0,0,0]},{"reg":36,"val":[0,0,0,0]},{"reg":0,"val":0},{"reg":127,"val":0}],"poweron":[{"reg":3,"val":3}],"poweroff":[{"reg":3,"val":0}]}

They all work as I hoped. Of course you can change some things and get everything coming out of the left speaker instead.

I haven't configured for mono (PBTL) mode yet and worked out those control sets, but that's next.

frdfsnlght, 2025-02-04:
And now I've enabled PBTL mode with a single speaker.

**Single speaker (PBTL mode), mono mix (L+R)/2:**
`{"init":[{"reg":3,"val":2},{"reg":3,"val":3},{"reg":2,"val":4},{"reg":0,"val":0},{"reg":127,"val":140},{"reg":0,"val":41},{"reg":24,"val":[0,64,38,231]},{"reg":28,"val":[0,64,38,231]},{"reg":32,"val":[0,0,0,0]},{"reg":36,"val":[0,0,0,0]},{"reg":0,"val":0},{"reg":127,"val":0}],"poweron":[{"reg":3,"val":3}],"poweroff":[{"reg":3,"val":0}]}
`

**Single speaker (PBTL mode), right input only:**
`{"init":[{"reg":3,"val":2},{"reg":3,"val":3},{"reg":2,"val":4},{"reg":0,"val":0},{"reg":127,"val":140},{"reg":0,"val":41},{"reg":24,"val":[0,67,218,96]},{"reg":28,"val":[0,0,0,0]},{"reg":32,"val":[0,0,0,0]},{"reg":36,"val":[0,0,0,0]},{"reg":0,"val":0},{"reg":127,"val":0}],"poweron":[{"reg":3,"val":3}],"poweroff":[{"reg":3,"val":0}]}
`

**Single speaker (PBTL mode), left input only:**
`{"init":[{"reg":3,"val":2},{"reg":3,"val":3},{"reg":2,"val":4},{"reg":0,"val":0},{"reg":127,"val":140},{"reg":0,"val":41},{"reg":24,"val":[0,0,0,0]},{"reg":28,"val":[0,67,218,96]},{"reg":32,"val":[0,0,0,0]},{"reg":36,"val":[0,0,0,0]},{"reg":0,"val":0},{"reg":127,"val":0}],"poweron":[{"reg":3,"val":3}],"poweroff":[{"reg":3,"val":0}]}
`

They all work perfectly. I can finally do what I've wanted to do. Thanks for the help!

anabolyc, 2025-02-04:
I need to double-check the labels, I may have mixed them. Please double check the sequence for both cases.

Right only: 
```
    { REG_BOOK_SET, 0x8c },
    { REG_PAGE_SET, 0x29 },
    { 0x18, 0x00 }, //  Input Mixer Left to left = -110 dB
    { 0x19, 0x00 },
    { 0x1a, 0x00 },
    { 0x1b, 0x00 },
    { 0x1c, 0x00 }, //  Input Mixer Right to left = 0 dB
    { 0x1d, 0x80 },
    { 0x1e, 0x00 },
    { 0x1f, 0x00 },
    { 0x20, 0x00 }, //  Input Mixer Left to right = -110 dB
    { 0x21, 0x00 },
    { 0x22, 0x00 },
    { 0x23, 0x00 },
    { 0x24, 0x00 }, //  Input Mixer Right to right = -110 dB
    { 0x25, 0x00 },
    { 0x26, 0x00 },
    { 0x27, 0x00 },
```

Left only:
```
    { REG_BOOK_SET, 0x8c },
    { REG_PAGE_SET, 0x29 },
    { 0x18, 0x00 }, //  Input Mixer Left to left = -110 dB
    { 0x19, 0x00 },
    { 0x1a, 0x00 },
    { 0x1b, 0x00 },
    { 0x1c, 0x00 }, //  Input Mixer Right to left = 0 dB
    { 0x1d, 0x80 },
    { 0x1e, 0x00 },
    { 0x1f, 0x00 },
    { 0x20, 0x00 }, //  Input Mixer Left to right = -110 dB
    { 0x21, 0x00 },
    { 0x22, 0x00 },
    { 0x23, 0x00 },
    { 0x24, 0x00 }, //  Input Mixer Right to right = -110 dB
    { 0x25, 0x00 },
    { 0x26, 0x00 },
    { 0x27, 0x00 },
```

Mono:
```
    { REG_BOOK_SET, 0x8c },
    { REG_PAGE_SET, 0x29 },
    { 0x18, 0x00 }, //  Input Mixer Left to left = -6 dB
    { 0x19, 0x40 },
    { 0x1a, 0x26 },
    { 0x1b, 0xe7 },
    { 0x1c, 0x00 }, //  Input Mixer Right to left = -6 dB
    { 0x1d, 0x40 },
    { 0x1e, 0x26 },
    { 0x1f, 0xe7 },
    { 0x20, 0x00 }, //  Input Mixer Left to right = -110 dB
    { 0x21, 0x00 },
    { 0x22, 0x00 },
    { 0x23, 0x00 },
    { 0x24, 0x00 }, //  Input Mixer Right to right = -110 dB
    { 0x25, 0x00 },
    { 0x26, 0x00 },
    { 0x27, 0x00 },
```

This is coming directly from TI's PurePath, including comments, so I'm confident about the channel labels. BTW, please check if your WS signal is not inverted by any chance

frdfsnlght, 2025-02-04:
Your left and right sequences look identical to me and your mono sequence is the same as what you originally provided. But I do see the value now for 0dB is 0x00 0x80 0x00 0x00 and not the value I was using, but that just changes the level a little. I'll correct my sequences with that value.

As for the WS signal, I have no way of checking that. Isn't that a signal between the ESP and the DAC? Do I have any control over that? I'm just using the config as it came shipped, except for changing the dac_controlset value.

anabolyc, 2025-02-04:
My mistake, left and right corrected. Mono is the same as above. 
Was can be inverted in the i2s init config, but if you didn't do any changes around it, it is probably not inverted by default.

frdfsnlght, 2025-02-04:
I didn't make that change. I was just looking at the squeezelite docs and I see that "ws" is defined there, but it doesn't look like it can be inverted like some of the other signals.

frdfsnlght, 2025-02-04:
I've updated my sequences above to change the 0dB level to 0x00 0x80 0x00 0x00 ([0,128,0,0]).

frdfsnlght, 2025-02-05:
Your idea about the WS signal being inverted may have merit, but I can't find a way to change it. I think it would explain a lot. However, looking at the ESP32 i2s library, specifically [this page](https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32/api-reference/peripherals/i2s.html), there doesn't seem to be a way to invert the WS line. But I'm wondering if some other field (maybe communication_format?) in the i2s_config_t structure needs to have a different value. I don't think squeezelite-ESP gives you the ability to tweak those fields though. I didn't dig enough into the squeezelite code to figure out what values are used in that structure, so I'm not sure if anything should be changed. But it was a thought.

anabolyc, 2025-02-05:
Search for `invert_flags` [here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/i2s.html)

frdfsnlght, 2025-02-05:
That looks interesting, but I can find it's use anywhere in the squeezelite code? And there's nothing in the squeezelite docs about the ability to invert the WS signal. So it might require a patch to squeezelite?

closed 2025-02-09 by anabolyc