sonocotta/esp32-audio-dock #28: HiFi-ESP ethernet

frdfsnlght, 2025-02-03:
I've just received my order for a HiFi-ESP board (rev F) with W5500 ethernet interface pre-installed. How do I get the interface working?

I've plugged in the antenna and got it working over WiFi, but I'd really like to use the ethernet port instead. I assume I have to configure the "SPI Bus Parameters" under the "Hardware" tab in squeezelite-ESP32, but I'm not 100% sure what put in the various fields.

Any help would be appreciated.

anabolyc, 2025-02-03:
Hey @frdfsnlght , please have a look [here](https://github.com/sonocotta/esp32-audio-dock?tab=readme-ov-file#optional-spi-ethernet-module)

frdfsnlght, 2025-02-04:
For some reason I didn't see that before. I'll give it a try tomorrow. Thanks.

frdfsnlght, 2025-02-04:
I've got the ethernet configured with the settings you've referenced. Works perfectly.

However, as I noted in issue #27, the right/left output is swapped just like in the Louder-ESP. Not a huge problem, I'll just swap the RCA plugs on the cable I'm using.

anabolyc, 2025-02-04:
Thanks for letting me know. I'll double check the labels

bib2, 2025-02-05:
Hi please can you explain me how i can do these changes on the louder esp32 to enable lan ethernet?

Do i have to flash it again with changed settings or is there a terminal available where i can enter this commands (e.g. like on raspberry)? Maybe Telnet?

anabolyc, 2025-02-05:
@bib2 
- Access your device over HTTP (you need to find it's IP from your router or from device serial console)
- Reboot to recovery (the button is in the update section I think)
- Access NVS settings when in Recovery
- Update spi and ethernet values as stated in docs
- Save and exit recovery

bib2, 2025-02-10:
I did this like described. But i can´t find it on my dhcp server.

Is the ap-ip-settings also used with ethernet? I can see 192.168.4.1 there.

I can connect with my smartphone over wifi, but i don´t see it ony my router menu.

If i add the two lines in the recovery mode, is dhcp activated or do i have to do something more?

anabolyc, 2025-02-10:
Squeezelite-esp32 is not able to use both ethernet and wifi at the same time. With the eth config in place, it will boot up and try to init ethernet and get IP from the DHCP server (or use static IP). If unsuccessful, it will reboot into wifi mode and no longer use ethernet, until the next reboot. That seems to be the case. 

Can you please check and confirm that after a reset ethernet logs are present and check if there is anything about DHCP. Are there any lights on the ethernet when it boots (greed/yellow next to RJ45 and red power light on the bottom)?

bib2, 2025-02-10:
Green led is on, orange is blinking. When i maked a reboot, i can´t access via the old ip. I open my smartphone wifi settings and there i can see the AP again. I have to enter the AP and login to the wifi again.

ping is not working. When i connect to the wifi, i get back the ping immediately.

anabolyc, 2025-02-10:
Okay, got it, that sounds promising. 

When it reboots and gets IP from the DHCP it will not be the same address, since the MAC of the wifi is different from ethernet. Best would be to attach a serial debugger over the USB-C (Don't connect high voltage at the same time!) and capture the boot log using the web-installer or serial monitor app. Once more, when ethernet is active, wifi will be completely off, so don't count on it to connect to the board, USB-c is your only friend

bib2, 2025-02-10:
Ok. But to clarify:

ip-settings in nve editor are for wifi only? The setting starts with ap- ... There are no ip settings for ethernet in nve editor?

Ethernet is DHCP only? No fixed IP?

And ap setting is for Access Point only? When i connect to a wifi, it will be useless? Then dhcp only?

anabolyc, 2025-02-10:
Please use  [web installer](https://sonocotta.github.io/esp32-audio-dock/) with the **erase** check box pressed. If it didn't help, I'd agree it is a hardware issue, I'll replace your board.