sonocotta/esp32-audio-dock #29: Unable to Flash Louder ESP32-S3

Dotsch2005, 2025-02-10:
I want to install the snap client via “https://sonocotta.github.io/esparagus-snapclient/” and connect the board.

Access via “Logs & Console” works without any problems.

Is there a trick?

The WLAN hotspot does not work either.

[esp-web-tools-logs (2).txt](https://github.com/user-attachments/files/18733840/esp-web-tools-logs.2.txt)

anabolyc, 2025-02-10:
You might need to reset the board with FLASH button pressed. Please try that. Also, if that didn;t help, please include the logs from the Chrom Dev Tools while you're trying to flash it

Dotsch2005, 2025-02-10:
Is it the TOO Button? 

[image]

Dotsch2005, 2025-02-10:
Okay, it looks like the ESP32-S3 board is not supported for Snapcast, or is it?

[image]

[sonocotta.github.io-1739199553254.log](https://github.com/user-attachments/files/18735631/sonocotta.github.io-1739199553254.log)

anabolyc, 2025-02-10:
Oh, I totally missed it. You're right, snapcast is not supported on the S3 at the moment. Some time ago I managed to build and flash it from the source code, but it was stuttering. I think it was a memory issue, not a DAC issue. Can you please reach out to [CarlosDerSeher](https://github.com/CarlosDerSeher/snapclient) and check if he's looking to add S3 support? I'm willing to give away a Louder-ESP32S3 board to anyone willing to make it work, shipping included.

Dotsch2005, 2025-02-10:
Is a program modification necessary here or must another sdkconfig be created?

I bought 3 boards of the Louder ESP32-S3, so a implementation would also be interesting for me.

I would have a look around [CarlosDerSeher](https://github.com/CarlosDerSeher/snapclient).

Dotsch2005, 2025-02-11:
Thank you, I have opened an issue at [CarlosDerSeher Issue 112](https://github.com/CarlosDerSeher/snapclient/issues/112)

I have tried to modify the SDKConfig myself and have also found the problem with the APLL. As you said, it's only a minimal change.

Do you still have the SDKConfig? The board always resets after HAL: INIT.

anabolyc, 2025-02-11:
Please look at [this branch](https://github.com/sonocotta/esparagus-snapclient/tree/features/s3-support)

Dotsch2005, 2025-02-12:
I tried to solve the problem in the issue at [CarlosDerSeher](https://github.com/CarlosDerSeher/snapclient/issues/112).

I couldn't find the information in the [branch ](https://github.com/sonocotta/esparagus-snapclient/tree/features/s3-support) either.

According to the [documentation](https://www.tindie.com/products/sonocotta/louder-esp32/), the I2S and I2C pins are different. Is the I2C_ADDR the same?

I have already tested the ESP-IDF example for an I2C scanner, but unfortunately it cannot find an address. I can't find a suitable SDKConfig for the ESP32S3 in the [Branch](https://github.com/sonocotta/esparagus-snapclient/tree/features/s3-support) either.

anabolyc, 2025-02-12:
You're right, I didn't commit the sdkconfig in the branch, my bad. Please check [this one](https://github.com/sonocotta/esparagus-snapclient/blob/master/sdkconfig.esparagus-echo-debug), it is not the same board (MAX98357 DAC), but it is for S3 and it was building and running as far as I remember. 

I referred to the branch since it has [build error fixes](https://github.com/sonocotta/esparagus-snapclient/commit/f1b51dadc574875517ab0e1a2d13d5d836475d80) that should give you a green build to start with.

You found the correct pinout right. I2C address is the same, but I2C pinout is different. Long story short, every pin is different. I'll try to create a build for Louder-ESP32S3 that runs and produces a sound, and I'll share it that way. Give me some time, please.

Dotsch2005, 2025-02-13:
Thanks for checking and updating the repo.

Unfortunately I get the following error when compiling:
```
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "snapclient" version: eac5495-dirty
-- Adding linker script D:/Github/esparagus-snapclient/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script D:/Github/esp-idf-v5.1.1/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script D:/Github/esp-idf-v5.1.1/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script D:/Github/esp-idf-v5.1.1/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script D:/Github/esp-idf-v5.1.1/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script D:/Github/esp-idf-v5.1.1/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script D:/Github/esp-idf-v5.1.1/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script D:/Github/esp-idf-v5.1.1/components/soc/esp32s3/ld/esp32s3.peripherals.ld
CMake Error at D:/Github/esp-idf-v5.1.1/tools/cmake/component.cmake:313 (message):
  Include directory
  'D:/Github/esparagus-snapclient/components/improv_wifi/Improv-WiFi-Library/src'
  is not a directory.
Call Stack (most recent call first):
  D:/Github/esp-idf-v5.1.1/tools/cmake/component.cmake:482 (__component_add_include_dirs)
  components/improv_wifi/CMakeLists.txt:4 (idf_component_register)

-- Configuring incomplete, errors occurred!
See also "D:/Github/esparagus-snapclient/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

I checked for ESP-IDF v5.1.5 and the branch [fix_20ms_chunk_size_bug](https://github.com/CarlosDerSeher/snapclient/tree/fix_20ms_chunk_size_bug) as recommended by [CarlosDerSeher](https://github.com/CarlosDerSeher). Here the compile works, but it hangs.

It gets stuck at `HAL: INIT` when I enable PSRAM and when PSRAM is disabled it gets stuck at `WIFI: wifi_init_sta finished`.

Is it possible that the board is defective or does not have PSRAM installed?

Many many thanks

anabolyc, 2025-02-13:
You need to sync submodules, it will fix the build.
PSRAM is built-in on module, I really doubt there is an issue with that, just double check the PSRAM settings in the SDKCONFIG

ataa, 2025-02-13:
> Is it possible that the board is defective or does not have PSRAM installed?

Use this to check for PSRAM : https://github.com/AdamJHowell/ESP32_CPU_Memory_Info

It will generate a report like this:

```
Source file: C:\ESP32_CPU_Memory_Info\ESP32_CPU_Memory_Info.ino
ESP32 Chip model: ESP32-D0WDQ6
  Revision: 3
  Core count: 2
  Chip ID: 381348
  Frequency: 240
  Cycle count: 1002386537
  SDK version: v4.4.2
Heap: 
  Total: 363804
  Used: 25300
  Free: 338504
  Largest block: 118772
  Minimum free since boot: 333112
Flash: 
  Total: 16777216
  Speed: 80000000
Sketch: 
  Size: 264336
  Free: 1310720
PSRAM: 
  Total: 4192139
  Used: 0
  Free: 4192139
  Largest block: 4128756
  Minimum free since boot: 4192139
Next printout in 20 seconds.
```

Dotsch2005, 2025-02-13:
> You need to sync submodules, it will fix the build. PSRAM is built-in on module, I really doubt there is an issue with that, just double check the PSRAM settings in the SDKCONFIG

Oh man... I forgot about that. I'm so sorry!
With the branch and the SDKCONFIG the compile works.

Unfortunately it doesn't work with the latest version ([fix_20ms_chunk_size_bug](https://github.com/CarlosDerSeher/snapclient/tree/fix_20ms_chunk_size_bug)). I'll tinker around with it a bit more.

> Use this to check for PSRAM : https://github.com/AdamJHowell/ESP32_CPU_Memory_Info
>
> It will generate a report like this:
>
> ```
> Source file: C:\ESP32_CPU_Memory_Info\ESP32_CPU_Memory_Info.ino
> ESP32 [...]

This is what I was looking for. Thank you!

closed 2025-04-09 by anabolyc