sonocotta/esp32-audio-dock #31: Louder ESP32-S3 dead on arrival?

elektrinis, 2025-02-14:
I bought one of these on tindie. Should be preinstalled with squeezelite and LAN port.
When I powered it via USB, wifi AP did not appear. Connected LAN - LEDs not blinking. So it's dead, or not flashed?

Tried to flash it via windows installer, but getting "Failed to execute 'open' on 'SerialPort'". After trying many times, if I pass this, I then get "Failed to initialize. Try resetting your device or holding the BOOT button while clicking INSTALL."
There's no boot button, but tried it with IO0 - no help. Also tried it per ESP's instructions: hold boot, press reset, release boot. Still no action.

What else can I do?
I'm an engineer, but I no longer have the time to hack things together, I bought it to use it.

ataa, 2025-02-14:
Have you tried powering it using the power terminal instead of USB-C? It will accept 5V to 26V.

anabolyc, 2025-02-14:
Hey @elektrinis . S3 is a bit different from the regular ESP32. It has a USB port for flashing and debugging purposes, but it only gets initialized when the board boots up with the right parameters. That said, first you need to make sure you connect to the right serial port, ti will show up in the Linux as ttyACMx, not really sure how this looks like in the Windows.

Second, flashed or not, S3 sometimes requires placing it into download mode for esp-web-tools to work, as I said S3 is a bit different there. And the sequence you mentioned is right, IO0 (aka FLASH) pressed, RESET clicked, FLASH released. Look up Serail port after that and web flasher should work.

If it spins forever, you might want to open Chrome Web tools and the console will give you a bit more information on how it is going. Sometimes it is a permissions issue, in most cases wrong serial port.

Just to be clear, I flash and test every board myself if you got it from Tindie, so it was working when it left my hands. I'm fairly confident it is an issue we can fix, but also I'm open to a return and refund if it is more than you're ready to invest. Please feel free to bail out any time

elektrinis, 2025-02-14:
I'd prefer to get it working, as there are really no alternatives.
Do you test them with USB or external power?
I've tested with phone's brick and with laptop's port.
Also laptop brick did not work, which you may want to take a look in future.

My unit is acting strangely when connected to laptop. Sometimes I get a virtual COM port, other times it's the same com port plus JTAG interface (with a yellow exclamation mark - no driver found).

elektrinis, 2025-02-14:
OK, I've fried my laptop. I supplied 19V to the board and plugged in USB. Laptop switched off and will not turn on. Battery re-plug did not fix it.
So I'm guessing the 19V were supplied in reverse to the laptop.
A sensible thing would be to add a diode or ideal diode on USB power.
Now I have a brick worth €1k and upcoming recovery of data.

anabolyc, 2025-02-14:
> I'd prefer to get it working, as there are really no alternatives. Do you test them with USB or external power? I've tested with phone's brick and with laptop's port. Also laptop brick did not work, which you may want to take a look in future.

I test them with USB only

> My unit is acting strangely when connected to laptop. Sometimes I get a virtual COM port, other times it's the same com port plus JTAG interface (with a yellow exclamation mark - no driver found).

I guess you're missing the drivers then

> OK, I've fried my laptop. I supplied 19V to the board and plugged in USB. Laptop switched off and will not turn on. Battery re-plug did not fix it.

I'm really sorry about that, that's why it says underneath not to do that very specific thing
[image]

I thought long and hard before doing those schematics, and the main reason is I don't want every single unit to lose voltage and consequently output power and risk cooking the diode (it should withstand up to 5A),  when it is fairly easy to develop/debug/test using cable only, and only connect full power when ready to deploy.

You pushed 19V into your USB port, which should cook the USB driver, hopefully nothing more. I hope there is a fuse that can be replaced and bring it back to life.

elektrinis, 2025-02-15:
Did not see the warning. It's good it's there. As for laptop, I think damage may be deep, down to dc-dc converter. Not going to attempt a fix. Expensive lesson.

I bought a new laptop, restored all data and tried again. Same results. Every time I plug it in, I get different behavior. One time I get COM3 (driver fully installed), other time I get COM4 + "Unknown device". Another time I get again COM3 + JTAG interface.

[image]
[image]

Different errors on web installer. One time I get "can't open port", other time install fails with suggestions about buttons. I do the button thing, but nothing changes.

I've tested 3 different USB cables, two laptops, different browsers and spent a lot of time on this. I believe it's defective.

I'd like a replacement. How can I contact you for further steps?

anabolyc, 2025-02-15:
I can replace it, no worries, just trying to avoid the case when the new one will do the same trick on you. As I said, the S3 is not the same as the classic ESP3, it works differently. 

With the web tools, can you please open the Chome Dev Console on the console log and try again your routine? This should give you more information on how the communication flow goes.

anabolyc, 2025-02-16:
Is this whole log you see? Improv Serial relates to Wifi-configuration, that is not that much relevant for initial flashing. Errors are not that promising, you might simply missing device drivers. Let me try this on my Windows laptop and I'll see if it works and if the log look any similar. If you can provide full log here, I'll have something to compare to

elektrinis, 2025-02-17:
Yes, this is full log, there is no more output. Tried it a few times.

anabolyc, 2025-02-17:
BTW I can replace it with the regular ESP32 version, to make things simpler for you. Is there a specific reason for you to use S3?

elektrinis, 2025-02-17:
I have just tested flashing my devboard with second USB port. It worked properly.
I'm sure there is nothing wrong on my end, with drivers etc. There is nothing else I can do.
Let me know your email so we can resolve this matter privately. Thanks.

anabolyc, 2025-02-17:
> I have just tested flashing my devboard with second USB port. It worked properly. I'm sure there is nothing wrong on my end, with drivers etc. There is nothing else I can do. Let me know your email so we can resolve this matter privately. Thanks.

Okay, got it. Please reach out at andriy@sonocotta.com. Sorry about the confusion.

anabolyc, 2025-02-18:
Replacement board has been shipped out

elektrinis, 2025-03-05:
Replacement board arrived and is working fine. Thank you.
Issue resolved.

closed 2025-03-05 by elektrinis