sonocotta/esp32-audio-dock #32: How to connect ST7789 on Louder-ESP32-S3 ! Problem to identify PIN !

nanostra, 2025-02-15:
Hello,

Hello, I purchased a Louder-ESP32-S3 with an external antenna on Tindie. 
I do not have an optional Ethernet port connected. 
I am using Squeezelite.  

**1. I managed to connect a KU-040 rotary encoder and make it work properly with Squeezelite** 

- CLK on GPIO 12 SPI CLK, 
- DT on GPIO 11 SPI MOSI, 
- SW on GPIO10  

`Squeezelite NVS Editor --> Rotary_config --> A=11,B=12,SW=10,volume,longpress`

For your information, a short press = Play/Pause, rotating left/right increases the volume.
A long press changes the mode, rotating left/right selects the previous/next track.

**2. I am now trying to connect a 2.8" ST7789 240x320 display, but I can't find some pins mentioned in your documentation, especially PIN 37 OLED SPI DC!!! Where is this PIN 37 located?  Similarly, I cannot find PIN RGBOUT 9 anywhere!**

**Could you please tell me where PIN 37 and RGBOUT 9 are located and confirm that my connection diagram is correct?**

Thank you for this very nice project, even though I’m struggling to identify the pins I need! 😊

```
GND  -->GND
VCC	  --> 3.3V
SCL (SPI CLK)	--> OLED SPI CLK GPIO 12 (Shared with encoder)
SDA (SPI MOSI)  --> OLED SPI MOSI GPIO 11 (Shared with encoder)
RST (Reset)        --> OLED RST GPIO 38
**DC (Data/Command) --> ???? GPIO 37**
CS (Chip Select)  --> OLED SPI CS GPIO 39
BL (Backlight) --> Not connected
```

[image]

[image]

anabolyc, 2025-02-15:
Hey @nanostra 
Pin 37 is blocked by PSRAM, I didn't route it out. Pin 9 is used by I2C. You can use pretty much any pin available as a test point on the front and back sides. I routed out all pins that are not booked yet.

nanostra, 2025-02-16:
I solved the problem with these explanations. I was hesitant to use these pins, as the ones available in the documentation seemed to be limited to those listed. Maybe you could clarify this point in the documentation.

What is the version of Squeezelite installed on your page https://sonocotta.github.io/esp32-audio-dock/ ?

There are quite a few bugs with Spotify artworks, causing the Louder-ESP32S3 to crash... or it might be a compilation issue.
I have opened a ticket on the Squeezelite GitHub, but maybe it's better to ask you instead.

🚨 Issue encountered

When Spotify sends an image URL (artwork) for the current track, the ESP32 attempts to download it but fails with the following error in the logs. Squeezelite then completely crashes, and I have to reboot and remove the artwork:1 option.

```
I (50365) cspot: requesting artwork https://i.scdn.co/image/ab67616d00001e029a494f7d8909a6cc4ceb74ac
I (50365) display: playing Chris Brown - Under The Influence
[00:05:13.786] D MercurySession.cpp:252: Executing Mercury Request, type SEND
E (50415) HTTP_CLIENT: No transport found

> E (50425) HTTP_CLIENT: Please enable HTTPS at menuconfig to allow requesting via https

W (50425) cspot: artwork error or too large 0
```
Thank you for your great work! 😊

anabolyc, 2025-02-16:
I will have a look. Seems to be fairly easy to fix. I was planing to update the code to the latest as well. Give me some time and I'll give you a build to test.

anabolyc, 2025-02-17:
@nanostra can you please try it out at https://sonocotta.github.io/esp32-audio-dock/
I've updated to the latest code and build with the following flags updated
```
#
# ESP HTTP client
#
CONFIG_ESP_HTTP_CLIENT_ENABLE_HTTPS=y
CONFIG_ESP_HTTP_CLIENT_ENABLE_BASIC_AUTH=y
CONFIG_ESP_HTTP_CLIENT_ENABLE_DIGEST_AUTH=y
```

nanostra, 2025-02-17:
Hello, @anabolyc,

Thank you for your feedback... But apparently, this did not solve the problem, and the Louder still crashes, I have to reboot it and remove the artwork option.

Another possibility would be to retrieve the artwork using an HTTP request instead of HTTPS...

**Example:**

`http://i.scdn.co/image/ab67616d00001e02c58b581c98eafab982391a6d`

Additionally, when I enable Telnet, I completely lose access to the Louder, and I have to reinstall the firmware to regain control.
For logs, I use Console-Log from your web page.

Similarly, if I disable Bluetooth, which is logically unnecessary on the ESP32-S3, I also lose access to the Louder and have to reinstall the firmware. But these are separate issues.

**Questions:**

Do you have documentation for compiling the firmware for the Louder-ESP32-S3?
**At least, do you have the specific parameters needed for it?**
I usually work with PlatformIO. Would it be easy to port/compile the project on this platform?

```
[00:00:57.427] I MercurySession.cpp:42: Received packet, command: 178
[00:00:57.431] D MercurySession.cpp:174: Received mercury packet
[00:00:57.460] I MercurySession.cpp:42: Received packet, command: 178
[00:00:57.464] D MercurySession.cpp:174: Received mercury packet
[00:00:57.795] I TrackPlayer.cpp:206: Playing
[00:00:57.797] I Shim.cpp:120: new track started <> => <de8bfbbe2281f3162ebb36b89523e4d624cc484f>
[00:00:57.809] I Shim.cpp:262: next track will play in 0 ms
[00:00:57.827] _output_frames:150 track start sample rate: 44100 replay_gain: 0
[00:00:57.885] I Shim.cpp:416: next track's audio has reached DAC (offset 6085)
[00:00:57.893] D MercurySession.cpp:252: Executing Mercury Request, type SEND
I (58949) cspot: requesting artwork https://i.scdn.co/image/ab67616d00001e02c58b581c98eafab982391a6d
I (58949) display: playing Marnik - Boyz In Paris (with VINAI)
[00:00:57.951] D MercurySession.cpp:252: Executing Mercury Request, type SEND
E (59029) esp-tls-mbedtls: No server verification option set in esp_tls_cfg_t structure. Check esp_tls API reference
E (59029) esp-tls-mbedtls: Failed to set client configurations, returned [0x8017] (ESP_ERR_MBEDTLS_SSL_SETUP_FAILED)
E (59039) esp-tls: create_ssl_handle failed
E (59049) esp-tls: Failed to open new connection
E (59049) TRANSPORT_BASE: Failed to open a new connection
E (59059) HTTP_CLIENT: Connection failed, sock < 0
W (59069) cspot: artwork error or too large 0
I2S abort!
```
[nvs_config_squeezelite_config_avec_ecran+artwork.txt](https://github.com/user-attachments/files/18830718/nvs_config_squeezelite_config_avec_ecran%2Bartwork.txt)

anabolyc, 2025-02-17:
S3 support is not official, I'm afraid. I will add to the repo sdkconfig, that I'm using, it builds and runs with the latest code, with certain limitations like you already discovered, lack of telnet and Bluetooth. 

If the https issue is not S3 specific, good chances it can get fixed if you report it to sqeezelite-esp32 official repo. 

I tried in the past to make it build in Platformio but gave up in the end. They use quite old esp-idf and it is a bit like wrestling every time. But feel free to go for it.

anabolyc, 2025-02-17:
[Here](https://github.com/sonocotta/esp32-audio-dock/tree/main/firmware/squeezelite-esp32) it is

nanostra, 2025-02-17:
There is a branch for the S3, but it seems abandoned...
https://github.com/sle118/squeezelite-esp32/tree/4.4-s3

This Squeezelite project doesn’t seem very active anymore...

Do you know of any other similar projects that support AirPlay, rotary encoder control, and screen integration?

anabolyc, 2025-02-17:
I think squeezelite-esp32 authors switched to [euphonium](https://github.com/muvox-io/euphonium), I'm not sure in which stage it is and if it supports screen options

anabolyc, 2025-02-17:
Look into ESPHome + Home Assitant maybe

nanostra, 2025-02-17:
Thank you very much. I will take a look to euphonium. 

> Philippe44 answer me... to say "It's a config error from the s3 build.. As I don't provide "official" s3 build, the author should change menuconfig to enable HTTPS " :-)

I use Home Assistant with more than 90 devices. However, for this use case, I don’t want to depend on Home Assistant because I don’t see much benefit in it, like in LMS...

I also have a 3D speaker design that I’m testing, in which I wanted to integrate the Louder-ESP32-S3, but I’m now realizing that I probably shouldn’t have chosen the S3.

I’ll keep you updated on my progress

nanostra, 2025-02-17:
Do you see the last answer from Philippe44 ?

https://github.com/sle118/squeezelite-esp32/issues/473

> Indeed, in ESP-TLS, you have to:
> [] Allow potentially insecure options
> [] Skip server certificate verification by default (WARNING: ONLY FOR TESTING PURPOSE, READ HELP)

philippe44, 2025-02-17:
BTW, we have not switched to euphonium, we collaborated with the dev of euphonium as he wrote cspot and we exchanged some piece of code. I've not been active on squeezelite-esp32 much these past month as life have decided otherwise. But the s3 branch is not stalled, it's not used as a separated branch. All s3 related code is now in the main branch, si it's a single steel of code. Only have I been lazy on the sdkcondig file ...

anabolyc, 2025-02-18:
@philippe44 thanks for correcting me out, I may have misunderstood that

@nanostra how can I test it without a screen, I want to try disabling TLS verification. Alternatively, I can give you steps to build and test with S3. It is fairly straightforward.

nanostra, 2025-02-18:
@anabolyc You can probably try with the NVS file provided earlier in the conversation... but without the screen connected, I'm not sure if it will go all the way. 

And yes, I'm interested in compiling it myself.

anabolyc, 2025-02-18:
Okay @nanostra , it is fairly simple
Assuming you pulled the repo and all the submodules. Copy `sdkconfig` that I've provided [here](https://github.com/sonocotta/esp32-audio-dock/tree/main/firmware/squeezelite-esp32)

Run docker container for build
```
docker run --rm -it -v .:/project -w /project -v /dev:/dev --privileged espressif/idf:v4.4.7
```

Inside a container install one dependency
```
pip3 install protobuf grpcio-tools
```
 
Prepare for the build, change the sdkconfig settings related to tls (Use / for search inside menuconfig)
```
cp sdkconfig.esp32s3 sdkconfig
idf.py set-target esp32s3
IDF_TARGET=esp32s3 idf.py menuconfig
```

Build and flash
```
IDF_TARGET=esp32s3 idf.py build flash monitor
```

closed 2025-04-09 by anabolyc