sonocotta/esp32-audio-dock #34: Translate C-lib to squeezelite dac_controlset

koriwi, 2025-02-19:
I think the title says everything.
i was thinking about the idea of creating an intermediate format, like json or yaml, and then run a script that can generate the C files or the `dac_controlset` for squeezelite. so you do not need to create those manually. Also you would not multiply your workload updating the same data in different formats.

I already looked at the header files and i think it should be doable. what is your opinion?

I really want to use a `louder` for subwoofer with squeezelite.

anabolyc, 2025-02-19:
I like the idea of using DSP settings with squeezelite-esp32. I don't think `dac_controlset` will work, it is fairly inefficient at storing binary data. The startup sequence can be up to ~1000 commands sent.

Some binary implementation would be much better, ideally integrated into LMS UI. I'd advise you to get in touch with squeezelite-esp32 developers and ask their opinion on this. I don't mind creating this work in a fork, but they would know the system better and probably have an idea of how this should be done.

koriwi, 2025-02-27:
Thx for your feedback. Maybe I can have a look this weekend!

closed 2025-08-09 by anabolyc