sonocotta/esp32-audio-dock #35: Louder-ESP32 "custom component" no longer supported in Home Assistant 2025.2

admn-monkey, 2025-03-02:
Will not validate or compile to update / install. Error below:

`
INFO ESPHome 2025.2.1
INFO Reading configuration /config/esphome/esphome-web-5ffc84.yaml...
Failed config

switch.custom: [source /config/esphome/esphome-web-5ffc84.yaml:60]
  
  The "custom" component has been removed. Consider conversion to an external component.
  https://esphome.io/guides/contributing#a-note-about-custom-components.
  platform: custom
  lambda: auto tas5805 = new TAS5805();
  App.register_component(tas5805);
  return {tas5805};
  switches: 
    name: Enable Amp
`

TommySharpNZ, 2025-03-02:
Yup was just about to log this too!

anabolyc, 2025-03-02:
Hey @admn-monkey @TommySharpNZ. Thanks for reporting. I'm working on this together with a few folks in the community. The test version is here:
```
substitutions:
  devicename: louder-esparagus-kantoor
  long_devicename: "Louder ESParagus kantoor"

esphome:
  name: "${devicename}"
  name_add_mac_suffix: false
  comment: "${long_devicename}"
  on_boot:
    priority: 250
    then:
    ## Set a volume limit just to be safe...
    - media_player.volume_set:
        id: louder_mediaplayer
        volume: 10%

esp32:
  board: mhetesp32minikit

wifi:
  ssid: !secret esphome_wifi_ssid
  password: !secret esphome_wifi_password
  ap:
    ssid: "$devicename Hotspot"
    password: !secret esphome_ap_password

captive_portal:
### Optional if you want ethernet (then remove all wifi config) ###
#ethernet:
#  type: W5500
#  clk_pin: GPIO18
#  mosi_pin: GPIO23
#  miso_pin: GPIO19
#  cs_pin: GPIO05
#  interrupt_pin: GPIO35
#  reset_pin: GPIO14

logger:
  level: DEBUG

api:
  encryption:
    key: !secret esphome_api_key

ota:
  platform: esphome
  password: !secret esphome_ota_password

psram:
  mode: octal
  speed: 80MHz

external_components:
  - source: github://mrtoy-me/esphome-components-test@main
    components: [ tas5805m ]
    refresh: 0s

i2c:
  id: i2c_component
  sda: GPIO21
  scl: GPIO27
  scan: True
  frequency: 400kHz
  timeout: 10ms

audio_dac:
  - platform: tas5805m
    id: tas5805_dac
    enable_pin: GPIO33

i2s_audio:
  i2s_lrclk_pin: GPIO25
  i2s_bclk_pin: GPIO26

media_player:
  - platform: i2s_audio
    name: "${long_devicename} mediaplayer"
    id: louder_mediaplayer
    dac_type: external
    i2s_dout_pin: GPIO22
    mode: stereo
    on_play:
      - switch.turn_on: enable_louder

switch:
  - platform: template
    name: "Enable ${long_devicename}"
    id: enable_louder
    optimistic: true
    turn_on_action:
      - lambda: !lambda |-
          id(tas5805_dac).set_deep_sleep_off();
    turn_off_action:
      - lambda: !lambda |-
          id(tas5805_dac).set_deep_sleep_on();

interval:
  - interval: 1min
    then:
      - if:
          condition:
            - switch.is_on: enable_louder
            - for:
                time: 30s
                condition:
                  or:
                    - media_player.is_idle: louder_mediaplayer
                    - media_player.is_paused: louder_mediaplayer
          then:
            - switch.turn_off: enable_louder
```

NickiNiclas, 2025-03-02:
Based on the above, here is my tested and working config for Louder-ESP32 **S3**

```
substitutions:
  devicename: louder-esp32-01
  long_devicename: "Louder ESP32"

esphome:
  name: "${devicename}"
  name_add_mac_suffix: false
  comment: "${long_devicename}"
  on_boot:
    priority: 800
    then:
    ## Set a volume limit just to be safe...
    - media_player.volume_set:
        id: louder_mediaplayer
        volume: 20%

external_components:
  - source: github://mrtoy-me/esphome-components-test@main
    components: [ tas5805m ]
    refresh: 0s

esp32:
  board: esp32-s3-devkitc-1
  variant: ESP32S3
  flash_size: 16MB
  framework:
    type: arduino

### Optional if you want ethernet (then remove all wifi config) ###
#ethernet:
#  type: W5500
#  clk_pin: GPIO12
#  mosi_pin: GPIO11
#  miso_pin: GPIO13
#  cs_pin: GPIO10
#  interrupt_pin: GPIO6
#  reset_pin: GPIO5

# Enable logging
logger:
  level: DEBUG
  

psram:
  mode: octal
  speed: 80MHz

audio_dac:
  - platform: tas5805m
    id: tas5805_dac
    enable_pin: GPIO33

switch:
  - platform: template
    name: "Enable ${long_devicename}"
    id: enable_louder
    optimistic: true
    turn_on_action:
      - lambda: !lambda |-
          id(tas5805_dac).set_deep_sleep_off();
    turn_off_action:
      - lambda: !lambda |-
          id(tas5805_dac).set_deep_sleep_on();

i2c:
  sda: GPIO8
  scl: GPIO9
  scan: True
  id: i2c_component

i2s_audio:
  i2s_lrclk_pin: GPIO15
  i2s_bclk_pin: GPIO14

interval:
  - interval: 1min
    then:
      - if:
          condition:
            - switch.is_on: enable_louder
            - for:
                time: 30s
                condition:
                  or:
                    - media_player.is_idle: louder_mediaplayer
                    - media_player.is_paused: louder_mediaplayer
          then:
            - switch.turn_off: enable_louder

media_player:
  - platform: i2s_audio
    name: "${long_devicename} mediaplayer"
    id: louder_mediaplayer
    dac_type: external
    i2s_dout_pin: GPIO16
    mode: stereo
    on_play:
      - switch.turn_on: enable_louder
### max_volume feature - uncomment below to enable
#     on_state:
#       then:
#         - if:
#             condition:
#               lambda: |-
#                 float current_volume = roundf(id(louderesp32).volume * 100) / 100.0;
#                 float max_vol = roundf(${max_volume} * 100) / 100.0;
#                 return current_volume > max_vol;
#             then:
#               - logger.log: "Volume exceeds max volume, resetting to max volume"
#               - media_player.volume_set:
#                   id: louderesp32
#                   volume: ${max_volume}
#               - logger.log: "Volume set to max volume"
```

TommySharpNZ, 2025-05-03:
I now get this error when trying to flash my rev H ESP Louder....

Octal PSRAM is only supported on ESP32-S3, not ESP32.

TommySharpNZ, 2025-05-03:
I'll grab the latest ESPHome code from the repository and see if that helps.

admn-monkey, 2025-05-12:
@anabolyc  That worked for me. Running well on ESPHome 2024.11.3 (ready now to switch to 2025.4.2, just not today).

admn-monkey, 2025-05-14:
@anabolyc I just tested with ESPHome Device Builder 2025.4.2 and also receive the error below. 

`  Octal PSRAM is only supported on ESP32-S3, not ESP32.
  mode: octal
  speed: 80000000.0
  enable_ecc: False`

anabolyc, 2025-05-21:
I'm testing the mrtoy's [dev branch](https://github.com/mrtoy-me/esphome-components-test/blob/dev/components/tas5805m/esp32S3_louder_idf_media.yaml) currently. With S3 it works flawlessly. With the ESP32 it crashes on playback. I'll try to play with SDKCONFIG settings

[image]

anabolyc, 2025-05-23:
While work is still in progress, I'm posting the config that works for me (external component work is not finalized yet). It is based on esp-idf and compared to the Arduino implementation, it works much better, no glitches and much more stable, at least based on my limited experience. Bonus points, this implements a 15-band equalizer, with -15..+15dB regulation, works really amazing, really looking forward to having it finalized

```yaml

substitutions:
  name: esphome-web-dc73f8
  friendly_name: louder-esp32
  task_stack_in_psram: "false"

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.11.0
  name_add_mac_suffix: false
  platformio_options:
    board_build.flash_mode: dio
  # on_boot:
  #   priority: 600  # After most components are initialized
  #   then:
  #     - lambda: |-
  #         id(tas5805m_dac).set_volume(int(id(current_volume)));
  #         ESP_LOGI("tas5805m", "Restored volume on boot: %.0f", id(current_volume));
esp32:
  board: mhetesp32minikit
  flash_size: 8MB
  framework:
    type: esp-idf
    version: recommended
    sdkconfig_options:
      CONFIG_BT_ALLOCATION_FROM_SPIRAM_FIRST: "y"
      CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY: "y"

      CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC: "y"
      CONFIG_MBEDTLS_SSL_PROTO_TLS1_3: "y"  # TLS1.3 support isn't enabled by default in IDF 5.1.5

# Enable logging
logger:
  level: DEBUG
  
# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
  platform: esphome
  password: !secret esphome_ota_password

wifi:
  ssid: !secret esphome_wifi_ssid
  password: !secret esphome_wifi_password
  ap:
    ssid: "$name Hotspot"
    password: !secret esphome_ap_password

captive_portal:

psram:
  speed: 80MHz

#  - source: /config/esphome/esp32-tas5805m-dac/esphome/components
#  - source: /config/esphome/external/esphome-components-test/components
external_components:
  - source: github://mrtoy-me/esphome-components-test@dev
    components: [ tas5805m ]
    refresh: 0s

# script:
#   - id: set_volume
#     parameters:
#       level: float
[... 868 lines cut ...]
              - media_player.is_playing:
              - lambda: return (!id(eq_gain_refreshed));
          then:
            - lambda: |-
                id(tas5805m_dac).refresh_eq_gains();
                id(eq_gain_refreshed) = true;

    turn_off_action:
      - lambda: id(tas5805m_dac).set_eq_off();

interval:
  - interval: 30s
    then:
      - if:
          condition:
            - switch.is_on: enable_louder
            - for:
                time: 120s
                condition:
                  or:
                    - media_player.is_idle: louder_mediaplayer
                    - media_player.is_paused: louder_mediaplayer
          then:
            - switch.turn_off: enable_louder
```

c-MM, 2025-05-29:
Thanks for the nice hardware, I received the Louder Esparagus this week and are playing around with it. I bought it to use as snapclient but tried this config and the equalizer is amazing. while searching around I found this pull request:
https://github.com/esphome/esphome/pull/8350
and tried to combine it with this config.
Unfortunatelly it fails to compile as soon as I add the audio_dac section:

> audio_dac:
  - platform: tas5805m
    id: tas5805m_dac
    enable_pin: GPIO33
    analog_gain: -9db

do you think it's possible to combine your work on the with the esphome snapclient support?

anabolyc, 2025-05-29:
Hey @c-MM , nice find. I didn't know snapclient was imported to ESPHome, need to give it a try. 

> do you think it's possible to combine your work on the equalizer with the esphome snapclient support?

I think it should be possible, although it is in active development still. What is the error you're getting? Can you quote also full yaml?

c-MM, 2025-05-29:
Hi @anabolyc, to be honest I don't know what I'm doing I don't have any experience with esphome except as a user.

What I did (and learned today) was the following:

```
git clone https://github.com/esphome/esphome.git
cd esphome
git fetch origin pull/8350/head:snaptest
git switch snaptest
./script/setup
. ./venv/bin/activate
ln -s ../louder-esp1.yaml .
esphome compile louder-esp1.yaml
```
It fails with:
```
Compiling .pioenvs/louder-esp1/src/esphome/components/socket/socket.cpp.o
src/esphome/components/snapclient/decoder.cpp: In function 'void esphome::snapclient::http_get_task(void*)':
src/esphome/components/snapclient/decoder.cpp:529:58: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'u_addr'; did you mean 'addr'?
  529 |     inet_pton(AF_INET, SNAPCAST_SERVER_HOST, &(remote_ip.u_addr.ip4.addr));
      |                                                          ^~~~~~
      |                                                          addr
src/esphome/components/snapclient/decoder.cpp:530:15: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'type'
  530 |     remote_ip.type = IPADDR_TYPE_V4;
      |               ^~~~
src/esphome/components/snapclient/decoder.cpp:600:33: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  600 |     hello_message.client_name = "libsnapcast";
      |                                 ^~~~~~~~~~~~~
src/esphome/components/snapclient/decoder.cpp:601:24: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  601 |     hello_message.os = "esp32";
      |                        ^~~~~~~
src/esphome/components/snapclient/decoder.cpp:602:26: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  602 |     hello_message.arch = "xtensa";
      |                          ^~~~~~~~
*** [.pioenvs/louder-esp1/src/esphome/components/snapclient/decoder.cpp.o] Error 1
In file included from /home/mm/.platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/components/driver/deprecated/driver/i2s.h:23,
                 from src/esphome/components/i2s_audio/i2s_audio.h:5,
                 from src/esphome/components/snapclient/snapclient.h:8,
                 from src/esphome/components/snapclient/snapclient.cpp:4:
/home/mm/.platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/components/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
/home/mm/.platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/components/driver/deprecated/driver/i2s.h:27:2: warning: #warning "This set of I2S APIs has been deprecated, please include 'driver/i2s_std.h', 'driver/i2s_pdm.h' or 'driver/i2s_tdm.h' instead. if you want to keep using the old APIs and ignore this warning, you can enable 'Suppress leagcy driver deprecated warning' option under 'I2S Configuration' menu in Kconfig" [-Wcpp]
   27 | #warning "This set of I2S APIs has been deprecated, \
      |  ^~~~~~~
src/esphome/components/snapclient/snapclient.h:25:22: error: 'audio_dac' has not been declared
   25 |   void set_audio_dac(audio_dac::AudioDac *audio_dac) { this->audio_dac_ = audio_dac; }
      |                      ^~~~~~~~~
src/esphome/components/snapclient/snapclient.h:25:42: error: expected ',' or '...' before '*' token
   25 |   void set_audio_dac(audio_dac::AudioDac *audio_dac) { this->audio_dac_ = audio_dac; }
      |                                          ^
src/esphome/components/snapclient/snapclient.h:44:3: error: 'audio_dac' does not name a type
   44 |   audio_dac::AudioDac *audio_dac_{nullptr};
      |   ^~~~~~~~~
src/esphome/components/snapclient/snapclient.h: In member function 'void esphome::snapclient::SnapClientComponent::set_audio_dac(int)':
src/esphome/components/snapclient/snapclient.h:25:62: error: 'class esphome::snapclient::SnapClientComponent' has no member named 'audio_dac_'; did you mean 'set_audio_dac'?
   25 |   void set_audio_dac(audio_dac::AudioDac *audio_dac) { this->audio_dac_ = audio_dac; }
      |                                                              ^~~~~~~~~~
      |                                                              set_audio_dac
src/esphome/components/snapclient/snapclient.h:25:75: error: 'audio_dac' was not declared in this scope; did you mean 'AudioDac'?
   25 |   void set_audio_dac(audio_dac::AudioDac *audio_dac) { this->audio_dac_ = audio_dac; }
      |                                                                           ^~~~~~~~~
      |                                                                           AudioDac
src/esphome/components/snapclient/snapclient.cpp: In member function 'virtual void esphome::snapclient::SnapClientComponent::setup()':
src/esphome/components/snapclient/snapclient.cpp:31:72: error: conversion from 'i2s_pin_config_t' to non-scalar type 'i2s_std_gpio_config_t' requested
   31 |   i2s_std_gpio_config_t i2s_pin_config0 = this->parent_->get_pin_config();
      |                                           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
[... cut ...]
                id(eq_gain_refreshed) = true;

    turn_off_action:
      - lambda: id(tas5805m_dac).set_eq_off();

#interval:
#  - interval: 30s
#    then:
#      - if:
#          condition:
#            - switch.is_on: enable_louder
#            - for:
#                time: 120s
#                condition:
#                  or:
#                    - media_player.is_idle: louder_mediaplayer
#                    - media_player.is_paused: louder_mediaplayer
#          then:
#            - switch.turn_off: enable_louder
```

anabolyc, 2025-05-30:
Wow, you went with building from the source, that's cool. I didn't do it myself, to be honest. I'd rather try to build it in ESPHome with components pulled from PR directly
```
external_components:
  - source: github://pr#8350
    components: [ snapclient ]
    refresh: 0s
``` 
Also, the code in the mrtoy's repo already went ahead, I'll update the yaml in a comment above a minute

anabolyc, 2025-05-30:
@c-MM you need this component also, not sure if it will allow to pull 2 PRs at the same time though
```   
  - source: github://pr#8898
    components: [ speaker, speaker/media_player ]
    refresh: 0s
```

although it looks like it failed to import tas5805m component from mrtoy's repo.

luar123, 2025-06-01:
> Upd.: can't get it to build, I think `hostname` is wrong. I tried both literal and ip address with the same result

You need to add
```
network:
  enable_ipv6: true
```
(another limitation of the current implementation)

c-MM, 2025-06-01:
Hi @anabolyc, I'm sorry I didn't see your trouble compiling it. Github seems only to send notification mails on new comments, not on edits.

anabolyc, 2025-06-01:
No worries, i was on the move all the weekend anyway

c-MM, 2025-06-02:
These are great news, so both versions work. Digital volume works for for me if you install the snapcast integration for Home Assistant. Just fun I added the ble_tracker and bluetooth_proxy to the config and it works even while playing audio over wifi.

ssurovich, 2025-06-16:
Im using the "working setup" - https://github.com/sonocotta/esp32-audio-dock/issues/35#issuecomment-2897511069

HA sees it without, but the Louder-ESP32 isnt acting "right" - If I crank up the volume on the device in HA and do a media player test, the volume is very low, close to not being able to hear it - but at the end, the volume usually spikes for the last second or two.   At times, the beginning volume is also loud for a second and then it drops to near silent?

Is this an issue with the HW or something the the build process?

ssurovich, 2025-07-27:
> Hey [@ssurovich](https://github.com/ssurovich), you can see in that config analog_gain: under audio_dac: is set to -9db. Valid values are -15db to 0db. If you set it to 0db, it will be louder but then you could experience clipping issues if your power supply voltage is not high enough. I'm using 0db with a 22V power supply and it works great!

Good point - I need to go back and try the new power supply I have for the ESP32S3 versions with the external antenna - needed those for some more distance and have 48V high Amp power supplied for those and they work great - but didnt try with the units I originally bought and posted about.  Thanks for the idea.