sonocotta/esp32-audio-dock #36: No sound when playing announcement

elektrinis, 2025-03-08:
I have a Louder ESP32s3 and it has a weird issue with playing my mp3 file.
I'm using it with squeezelite firmware (the one it shipped with), LMS and Home Assistant with Piper addon (text to speech).
In HA, I use it mainly as internet radio player, which works really well for a few days now.
I also use it to play announcements. For example, this one:
https://www.dropbox.com/scl/fi/lhd7rvgkuid3l50u5mwbs/someone-at-the-door.mp3
When sent to play, the status of player changes to "playing" and "media_position" advances until it finishes. Everything looks normal as if it was working fine, however there is no audio output, it's completely silent. It plays normally when switched back to internet radio stream.
If I visit the full url of "media_content_id", my browser plays the file normally.
Before this I was running original squeezelite firmware on chinese A1S devboard and it worked fine.
Any ideas?

anabolyc, 2025-03-08:
Hey @elektrinis 

First of all, S3 support of squeezelite-esp32 is unofficial, so I would be ready for some bugs. Second, would you be able to pull some logs over Serial.

I'm working on the updated TAS5805M driver that will implement a few diagnostics and fault-reading methods, I'm planning to start testing it in the next few weeks. I might be able to give you a better logging output soon, but for now if you can pull what you already have, that might be a good first step

elektrinis, 2025-05-21:
Just a short update I still have this issue and could not find a workaround.
Any news on diagnostics? What's the latest version and how to update without connecting USB?

anabolyc, 2025-05-21:
Hey @elektrinis . I'm currently testing a new esp-idf implementation with ESPHome, which implements enforced encoding before playback. Probably the problem lies there. When I finish testing for ESPHome, I can try out the same using squeezelite-esp32 with Piper or Google TTS (if I figure out how to do it). If you can find a way to enforce a specific media format, that would probably be the way to fix it.
Myself I was too lazy to fix it, so I deployed another ESPHome device just for text announcements. But that's me, I have a few extra boards and speakers I can spare.

StaRky33, 2026-11-07:
This is an old issue but I don't think it was fixed. On my side using a Loud-ESP32S3 the issue is very clear.
Case 1 :
1. Music is playing correctly.
2. I play an annoncement with Piper directly or through Music assistant.
3. The annoncement pause the music, plays and everything is fine.
4. The music resume normally

Case 2 : 
1. No music is playing. The DAC is on Mute.
2. I play an annoncement with Piper directly or through Music assistant.
3. The player is playing the sound but the DAC is still on Mute. It looks like nothing is happening.

Case 3 : 
1. No music is playing. The DAC is on Mute.
2. I mannually turn the DAC Mute off.
3. I play an annoncement with Piper directly or through Music assistant.
4. The annoncement is played correctly.
5. The DAC Mute is turned back On.

https://github.com/sonocotta/esp32-audio-dock/blob/main/firmware/esphome/packages/sendspin.yaml?utm_source=chatgpt.com

I think the "on_annoncement" section of the sendspin.yaml is missing an unmute when no music is played. This is probably a very easy fix.