sonocotta/esp32-audio-dock #39: Question - Mono Mode in Home Assistant possible

ms-ku, 2025-04-08:
Hello 
is it possible to activate the Mono/PBTL Mode on a Louder Esp32-S3 in Home Assistant?
Thanks and greetings
msku

anabolyc, 2025-04-09:
Hey @ms-ku are you using header files that I provide or working with [new external component](https://github.com/sonocotta/esp32-audio-dock/issues/35) from mrtoy?

ms-ku, 2025-04-09:
Hello
I'm not quite sure what you mean but at the moment i use the yaml example from mrtoy-me
( https://github.com/mrtoy-me/esphome-components-test/blob/main/components/tas5805m/esp32_louder_idf_media.yaml ).
The Louder ESP32-S3 ia a newer one version (deliverd last week).
Thanks and greetings
msku

anabolyc, 2025-04-09:
At this point init sequence is coded into the [tas5805m_config_2.0+basic.h](https://github.com/mrtoy-me/esphome-components-test/blob/main/components/tas5805m/tas5805m_config_2.0%2Bbasic.h) file, so there is no quick way to change it. When the component work is done, it will be a yaml property that you can set in the config.

ms-ku, 2025-04-09:
Hello
Thanks for the info.
Greetings
msku

closed 2025-08-10 by anabolyc