sonocotta/esp32-audio-dock #40: Stream running but no audio output on LouderESP S3

finally-fancy, 2025-05-10:
Hi there,

I'm using the **LouderESP S3** board. Everything seems to initialize correctly — the amp powers on via I2C, WiFi connects, the MP3 stream is fetched without any errors, and `audio.isRunning()` returns `true`. But: no sound is heard from the speaker.

I'm powering the LouderESP via **12V through the barrel jack** (to ensure the amplifier has proper power), while simultaneously connected via **USB-C** for serial monitoring.

Here's the minimal sketch I'm using for testing:

```cpp
#include <Arduino.h>
#include <Wire.h>
#include <WiFi.h>
#include <Audio.h>

const char* ssid = "mySSID";
const char* pw   = "########";

const int AMP_PWDN = 17;
const uint8_t AMP_I2C = 0x2D;

Audio audio;

void amp_on() {
  pinMode(AMP_PWDN, OUTPUT);
  digitalWrite(AMP_PWDN, LOW);
  delay(5);

  Wire.begin(8, 9);
  Wire.beginTransmission(AMP_I2C); Wire.write(0x00); Wire.write(0x00); Wire.endTransmission();
  Wire.beginTransmission(AMP_I2C); Wire.write(0x02); Wire.write(0x00); Wire.endTransmission();
  Wire.beginTransmission(AMP_I2C); Wire.write(0x03); Wire.write(0x00); Wire.endTransmission();
}

void setup() {
  Serial.begin(115200); delay(5000);
  amp_on();

  WiFi.begin(ssid, pw);
  while (WiFi.status() != WL_CONNECTED) delay(100);

  audio.setPinout(PIN_I2S_SCK, PIN_I2S_FS, PIN_I2S_SD);
  audio.setVolume(21);
  audio.connecttohost("http://st03.dlf.de/dlf/03/128/mp3/stream.mp3");

  if (audio.isRunning()) {
    Serial.println("Audio stream is running");
  } else {
    Serial.println("Audio stream failed to start");
  }
}

void loop() {
  audio.loop();
}
```

my platformio.ini
```
[env:s3dev]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
monitor_speed = 115200

build_flags =
    -DARDUINO_USB_CDC_ON_BOOT=1
    -DCORE_DEBUG_LEVEL=5
    -DBOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue
    -D PIN_I2C_SDA=8
    -D PIN_I2C_SCL=9
    -D PIN_I2S_FS=15
    -D PIN_I2S_SCK=14
    -D PIN_I2S_SD=16
```

finally-fancy, 2025-05-10:
**Additional note:**
I had 8 Ω speakers connected and powered the board via the 12 V barrel jack. Later, I noticed the datasheet says:

> **“2×25 W (8 Ω, 1% THD+N) at 22 V”**

Does this mean that I *need* to supply 22 V to get any audible output? The barrel jack is labeled "5 V–26 V", so I assumed 12 V would be sufficient — but perhaps that's too low for meaningful audio output with 8 Ω speakers?

anabolyc, 2025-05-10:
First of all, please make sure your board do not have a warning on the back
about simultaneous power from usb and barrel jack, you may fry tour usb
port (earlier revisions)

Second, you get max power with 26v as noted in specs, but it will work
starting from 5v with lower power output.

Can you please cross check with examples in this repo, I don’t think i2c
communication in your snippet is right. Linked repo has minimal esp-idf
code that should give you quick start

finally-fancy, 2025-05-10:
There are two power input available, the one with the barrel jack and the other one with +5V (and the warning). I'm using the the barrel one to provide power to the amplifier(?).

All the samples do not run. I get always a `panic` in the log..

```
 *  Executing task: platformio run --target upload --target monitor --environment louder-esp32s3 

Processing louder-esp32s3 (board: lolin_s3_mini; framework: arduino; platform: espressif32)
----------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/lolin_s3_mini.html
PLATFORM: Espressif 32 (6.10.0) > WEMOS LOLIN S3 Mini
HARDWARE: ESP32S3 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- esp32-tas5805m-dac @ 0.0.0+20231111144526.sha.aa6b7a1
|-- I2S @ 1.0
Building in release mode
Retrieving maximum program size .pio/build/louder-esp32s3/firmware.elf
Checking size .pio/build/louder-esp32s3/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   6.0% (used 19768 bytes from 327680 bytes)
Flash: [===       ]  30.5% (used 399673 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: /dev/cu.usbmodem2101
Uploading .pio/build/louder-esp32s3/firmware.bin
esptool.py v4.5.1
Serial port /dev/cu.usbmodem2101
Connecting...
Chip is ESP32-S3 (revision v0.2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 94:a9:90:0c:40:20
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00071fff...
Compressed 15104 bytes to 10430...
Writing at 0x00000000... (100 %)
Wrote 15104 bytes (10430 compressed) at 0x00000000 in 0.3 seconds (effective 431.6 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 367.0 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 555.4 kbit/s)...
Hash of data verified.
[... 149 lines cut ...]
      (inlined by) linenoiseEdit at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/console/linenoise/linenoise.c:847
      (inlined by) linenoiseRaw at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/console/linenoise/linenoise.c:1104
      (inlined by) linenoise at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/console/linenoise/linenoise.c:1156
  #84 0x42003f0e in CommandLine::process_serial_commands() at .pio/libdeps/louder-esp32s3/esp32-tas5805m-dac/include/commandline.hpp:52
  #85 0x42004357 in CommandLine::loop() at .pio/libdeps/louder-esp32s3/esp32-tas5805m-dac/include/commandline.hpp:173
      (inlined by) loop() at src/main.cpp:53
  #86 0x420082b1 in loopTask(void*) at /Users/D023280/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50

ELF file SHA256: e30945d8250333c7

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42034812
  #0  0x42034812 in esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_pm/pm_impl.c:855

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
[   123][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
```

finally-fancy, 2025-05-10:
I have the Louder ESP S3 11/24 Rev J2

anabolyc, 2025-05-10:
Please do not connect 2 sources at the same time, you WILL fry your laptop.
I’ll try to fix examples when i get back to the office on Monday. Can you
please try looking into the library code, specifically i2c and i2s startup
code  implementation

anabolyc, 2025-05-11:
@finally-fancy, you're right, there is a stack overflow in the commandline library, that is unrelated to the audio but causes a crash on the S3. I'll try to fix it tomorrow.

finally-fancy, 2025-05-13:
Just to be safe, I also ordered two of the "regular" ESP32 (non-S3) boards. They seem to cause fewer issues and actually offer more features. I'm still not sure what the specific use case for the S3 is supposed to be — doesn’t seem very clear to me. Probably my mistake.

anabolyc, 2025-05-13:
@finally-fancy In the ideal world S3 is more modern ship that offers
- more and faster PSRAM
- slightly faster CPU
- more GPIOs
- (fancy) debugging over USB with breakpoints and memory access at no cost
- but lacks Bluetooth for reasons unknown

In reality though fortware support matters most, and simple fact is, old ESP32 wins it

finally-fancy, 2025-05-15:
Just checking in to see if you’ve had a chance to look into this issue or if there’s any update.

anabolyc, 2025-05-15:
Hey @finally-fancy . Sorry, I've been crazy busy this week. The code will stop crashing on the S3 when you comment [this single line](https://github.com/sonocotta/esp32-audio-dock/blob/01df94ae348d6beb1a3bed5924fcab674ba90a22/firmware/esp32-i2s-web-radio/src/main.cpp#L58), but obviously console client will stop to work. I found this to have something to do with the way USB serial is processing input data differently than UART serial, but didn't manage to make the fix compile yet. If you can, please look at the way audio initialisation is done, since console client is there more for your convenience, and has nothing to do with the DAC

closed 2025-08-11 by anabolyc