sonocotta/esp32-audio-dock #42: Hifi ESP32 S3: No programmable mute control

davidandreoletti, 2025-06-26:
Hi @sonocotta 

On [Schematics](https://github.com/sonocotta/esp32-audio-dock/blob/e09d5e691483df712a825f8a4eba469b99183b4a/hardware/5-hifi-esp32s3/2409-hifi-esp-g-schematic.pdf):
- PCM5100's mute pin is XSMT.
- XSMT pin seems  physically connected to 3V3 rail on the board. 

When:
- XSMT is low, audio is muted
- XSMT is high, audio is unmuted audio. 

Hence, when the board is powered on, the PCM5100 is unmuted.

I would like to control when to mute while the Hifi ESP32 S3 board is powered on, what's the recommended approach ?

anabolyc, 2025-06-26:
Hey @davidandreoletti, unfortunately, I didn't implement pin control on the DAC, since it is typically very low noise even when un-muted and no audio feed in.
In your case, I'd advise using PAM8046 mute pin, since this is the place where amplification happens, so it will be responsible for most of the noise (even if you shunt the amp input altogether)

Another consideration, how much voltage are you planning to supply to the PAM8046? Please check what the voltage swing amp expects on the input. Typically, 2.1V RMS of PCM5100 DAC needs a voltage divider before it enters the amp, and the lower leg of it would shunt the amp input when the DAC is off, which should prevent the noise from the DAC once more.

davidandreoletti, 2025-06-26:
@anabolyc

>  advise using PAM8046 mute pin

Got it. Thank you.

> how much voltage are you planning to supply to the PAM8046?

PAM8406's [datasheet](https://docs.rs-online.com/7c06/0900766b81268cf8.pdf) states:
- supply voltage: 2.5V to 5.5V
- input voltage  is stated as -0.3 to VDD+0.3V, which I understand as -0.3 to 5.8V.
- with 4ohm/5W speaker, graph 17 states 5V supply voltage yields 5V 5000mW of output power

PCM5100's [datasheet](https://www.ti.com/lit/ds/symlink/pcm5100.pdf?ts=1750948462300&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FPCM5100) states:
-- Full scale output: 2.1V RMS  (ie this output will be AMP's input voltage).
-- Output Logic Level High (0.8*VDD or 0.8*3.3V=2.64V) where 3.3V is the voltage supplied to the PCM5100 by the "Hifi ESP32 S3 board"
-- Output Logic Level Low (0.22*VDD or 0.22*3.3V=0.72V) 

So I intend for the PAM8406:
- to receive a 5V supply voltage for each 4ohm/5W speakers to be driven to 5W. 
- to not have any voltage divider between the DAC's output voltage and the AMP's input voltage since (PAM8406) -0.3V < (PCM5100) 0.72V < (PCM5100) 2.1V RMS < (PCM5100) 2.64V < (PAM8406) 5.8V

Sounds good ?

> the lower leg of it would shunt the amp input when the DAC is off, which should prevent the noise from the DAC once more.

Would you mind explainning "lower leg of it" in simpler terms please ?

anabolyc, 2025-06-26:
The input voltage indicated absolute maximum input value would not damage by this amp, but it will be heavily distorted if you choose to drive it with a 5V+ signal. Based on the datasheet, gain is 24db (I think), which is around 16x amplification. So it would expect around 0.2..0.3V on the input. If you connect it to PCM5102 without a voltage divider, you should expect heavy distortions (although it would not damage it).  

Logic Levels of the PCM5100 are irrelevant here, sice this applies to analog section only.

> Would you mind explainning "lower leg of it" in simpler terms please ?

What you're looking to do

[image]

If you imagine that PCM5100 output would go floating for some reason (DAC lose power), the lower leg of the voltage divider would shunt the amp input to the ground with 10K, which should be enough to keep it quiet. You can pick different values, of course, just try to find the right ratio to not drive the amp to clipping

closed 2025-08-24 by anabolyc