sonocotta/esp32-audio-dock #47: Crackling sound for Louder-ESP32S3 on esphome

XtracT, 2025-08-24:
Hello,

I am trying to set up my Louder board using Esphome, but I hear a continuous crackling sound with it. It does not happen when I flash squeezelite-esp32 though. Any ideas how to fix it? I am using the yaml from the repo directly, no modifications. 

BR;

/Xtract

anabolyc, 2025-08-24:
Hey @XtracT 

I never had this issue, by mrtoy mentioned it as well. Can you please try it with the following change:

```yaml
media_player:
  - platform: speaker
    name: Mediaplayer
    id: louder_mediaplayer
    buffer_size: 100000
```

XtracT, 2025-08-24:
So I tried again this morning, flashing from squeezelite-esp32 to esphome again, and the crackling is gone. I do not know if it could have been some configuration problem (although I am pretty sure I completely erased the device) or the dac not being reset or something, maybe I did not power cycle. In any case, now it works as it should, I was having some issues with squeezelite when I was changing playlists back and forth, but now with esphome it is a simpler setup and it works for my application. 

Thanks for the quick response!

closed 2025-08-24 by XtracT

anabolyc, 2025-08-24:
Glad it worked out well )